wasm-micro-runtime/wasm-micro-runtime #2667: Wrong output of i32.rotl in llvm-jit mode

unicornt, 2023-10-24:
### Test Case

[case.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13114528/report14.zip)

### Result

The second output in `llvm-jit` mode is  `1536` while the right answer is `-2`.
[image]

### Extra Information

This is a case I extra from other file. The key part is `line 54-71` of `case.wat` file.
[image]

`function print` is a print function which needs a i32 for parameter and return the same i32 as result.
`function print` will print the given i32 variable to the stdout.

The different output between `llvm-jit` mode and `fast-jit` mode means the result of `i32.rotl` is wrong in `llvm-jit` mode.

By the way, if I uncomment `line 62` in `case.wat` file, the output will be the same. So I guess maybe it is also a wrong optimization in `llvm-jit` mode.

### Versions and Environment

wamr version : 1.2.3

Operating system: Ubuntu 20.04

Architecture: x86_64

wenyongh, 2023-11-01:
Hi, thanks for reporting the issue! Please try #2697.

unicornt, 2023-11-01:
> Hi, thanks for reporting the issue! Please try #2697.

Yes, it really works. Could you please tell me how did you fix it? Is it about `mask` when calculating `Calculate (bits - shift_count) & mask`?

wenyongh, 2023-11-01:
Yes, the formula should be like the implementation of interpreter:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/interpreter/wasm_interp_classic.c#L122

But here it didn't `and` the `mask` for `(bits - shift_count)`, so the actual shift bits for `>>` may be invalid, e.g. 32 but not 0.

unicornt, 2023-11-01:
Thank you for the explanation!

closed 2023-11-01 by unicornt