wasm-micro-runtime/wasm-micro-runtime #2668: Maybe Create directorry bug?

orangeC23, 2023-10-24:
### Steps to Reproduce
(1) The cfile is :
```
#include <stdio.h>
#include <stdlib.h>
#include <fcntl.h>
#include <sys/stat.h>
#include <sys/types.h>
#include <unistd.h>
#include <string.h>

int path_create_directory(int fd, const char *base_directory, const char *new_directory, mode_t right){
    printf("Enter path_create_directory.\n");

    if (mkdirat(fd, new_directory, right) == -1) {
        perror("Error creating new directory");
        close(fd);
        return 1;
    }

    printf("Directory '%s' created under '%s'\n", new_directory, base_directory);
    printf("Leave path_create_directory.\n");
    return fd;
}

int snapshot(int myfd){   
    printf("Enter snapshot\n");
    
    struct stat file_info;
    if (fstat(myfd, &file_info) == -1) {
        perror("Error getting file attributes");
        close(myfd);
        return 1;
    }

    printf("File Size: %lld bytes \n", (long long)file_info.st_size);
    
    if (close(myfd) == -1) {
        perror("Error closing file");
        return 1;
    }

    printf("Leave snapshot\n");
}

int get_fd(const char* file_name, int open_style){
    int fd = open(file_name, open_style);
    if (fd == -1) {
        perror("Failed to open the file");
        return 1;
    }

    return fd;
}

int path_create_directoryZOyjKVxmtB (int fd, const char *base_directory){
    const char *new_directory = "newdir";
    mode_t right = 0200;
    return path_create_directory(fd, base_directory, new_directory, right);
}

int main() {
    const char* file_name = "Data/mydir";
    int open_style= O_RDONLY;
    int fd = get_fd(file_name, open_style);
    path_create_directoryZOyjKVxmtB(fd, file_name);
    snapshot(fd);
    return 0;
}

```
(2)compile the c file into wasm: `./wasi-sdk-16.0/bin/clang --target=wasm32-unkown-wasi --sysroot=./wasi-sdk-16.0/share/wasi-sysroot createdir.c -o createdir.wasm`
(3)exeute open.wasm
`iwasm  --dir=./Data createdir.wasm`
The permission of Data/mydir is 0700, user1 create the Data/mydir directory before and user1 execute the Wasm file.

### Expected Results
Using `gcc createdir.c -o createdir` and `./createdir` to execute get the following result:
```
Enter path_create_directory.
Directory 'newdir' created under 'Data/mydir'
Leave path_create_directory.
Enter snapshot
File Size: 4096 bytes 
Leave snapshot
```
And using 'ls -al' to check the newdir, the result is -w-------,this is the same as the C source code setting "0200".

### Actual Results
wamr also prints:
```
Enter path_create_directory.
Directory 'newdir' created under 'Data/mydir'
Leave path_create_directory.
Enter snapshot
File Size: 4096 bytes 
Leave snapshot
```
However, using 'ls -al' to check the newdir, the result is drwxr-xr-x, which is different from the C source code setting "0200".
I'm not sure whether this could be a bug. Maybe a bug or something different?
Thanks a lot !

### Versions and Environment
WAMR-1.2.3
Operating system: Ubuntu 20.04

Architecture: x86_64

lum1n0us, 2023-10-24:
It is not a bug. wasi-libc will use `__wasilibc_nocwd_mkdirat_nomode()` to implement `mkdirat()`. As its name, `__wasilibc_nocwd_mkdirat_nomode()` will drop `mode`.

orangeC23, 2023-10-25:
Thanks for your reply @

closed 2023-10-25 by orangeC23