wasm-micro-runtime/wasm-micro-runtime #2677: wamr encountered an error during AOT execution

XinyuShe, 2023-10-28:
### Test Case
[filea8.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13194836/filea8.zip)
### Result
```
==================================================
/home/exp/simd/filea8.wasm
-----------------wamr-----------------
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file /home/exp/simd/filea8.wasm.aot was generated.
Aborted

-----------------wasmedge-----------------
1324035698926385749195016455103447040
1682754050
1324035698926385749195016455103447040
1324035698926385749195016455103447040
1324035698926385749195016455103447040
1324035698926385749195016455103447040
1324035698926385749195016455103447040

==================================================
```
I ran filea8.wasm with wamr and wasmedge, wamr aborted
### Versions and Environment
```
root@2f9b6c2c3e26:/home/exp# wamrc --version
wamrc 1.2.3
root@2f9b6c2c3e26:/home/exp# iwasm --version
iwasm 1.2.3

Operating System: Ubuntu 22.04.3 LTS (GNU/Linux 5.15.0-79-generic x86_64)
```

wenyongh, 2023-11-01:
Hi, thanks for reporting the issue, please try the patch and re-compile the wamrc:
```diff
diff --git a/core/iwasm/compilation/aot_emit_memory.c b/core/iwasm/compilation/aot_emit_memory.c
index 7484d4b5..8c35c3fe 100644
--- a/core/iwasm/compilation/aot_emit_memory.c
+++ b/core/iwasm/compilation/aot_emit_memory.c
@@ -157,7 +157,10 @@ aot_check_memory_overflow(AOTCompContext *comp_ctx, AOTFuncContext *func_ctx,

         if (mem_offset + bytes <= mem_data_size) {
             /* inside memory space */
-            offset1 = I32_CONST((uint32)mem_offset);
+            if (comp_ctx->pointer_size == sizeof(uint64))
+                offset1 = I64_CONST((uint32)mem_offset);
+            else
+                offset1 = I32_CONST((uint32)mem_offset);
             CHECK_LLVM_CONST(offset1);
             if (!enable_segue) {
                 if (!(maddr = LLVMBuildInBoundsGEP2(comp_ctx->builder,
```

wenyongh, 2023-11-01:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2697

XinyuShe, 2023-11-01:
Hi, I used this patch and recompiled the iwasm. 
The result is: iwasm successfully executed filea8.wasm without aborting. However, the execution results still differ from other runtimes.
[image]

wenyongh, 2023-11-01:
Hi, not sure which tool you used to convert decimal v128 to hex 128? I used python, it seems ok:
[image]

BTW, had better apply patch https://github.com/bytecodealliance/wasm-micro-runtime/pull/2697 before testing, I fixed several issues with it.

XinyuShe, 2023-11-01:
Apologize for the basic mistake I made and I was using an online binary conversion tool . 
And thanks very much for your reply and suggestion.

wenyongh, 2024-01-05:
Close this issue as it was fixed.