wasm-micro-runtime/wasm-micro-runtime #2678: Different output when running the same wasm file

XinyuShe, 2023-10-28:
I constructed a wasm binary with an exported function called "main". I used the following command and compared the outputs of WAMR with other runtimes. 
The results of WAMR and WASMEDGE are the same ,but different from others. I don't if it is a bug.

The wasm binary is randomly generated, so there isn't related source code
[filea434.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13196208/filea434.zip)

```
wamrc -o file434.wasm.aot file434.wasm
iwasm --heap-size=0  -f main file434.wasm.aot
```

The output of WAMR is as follows:
```
-----------------wamr-----------------
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file /home/obfuscator/simd/filea434.wasm.aot was generated.
<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128,<0xffffffffffffffff 0xffffffffffffffff>:v128
```

While the outputs of the other runtimes like wasmtime and wasmer are:
```
-----------------wasmtime-----------------
warning: using `--invoke` with a function that returns values is experimental and may break in the future
339950060001143025775386386672551723008
339950060001143025775386386672551723008
339950060001143025775386386672551723008
339950060001143025775386386672551723008
339950060001143025775386386672551723008
339950060001143025775386386672551723008
339950060001143025775386386672551723008
339950060001143025775386386672551723008
339950060001143025775386386672551723008

-----------------wasmer-----------------
339950060001143025775386386672551723008 339950060001143025775386386672551723008 339950060001143025775386386672551723008 339950060001143025775386386672551723008 339950060001143025775386386672551723008 339950060001143025775386386672551723008 339950060001143025775386386672551723008 339950060001143025775386386672551723008 339950060001143025775386386672551723008

-----------------wasmedge-----------------
340282366920938463463374607431768211455
340282366920938463463374607431768211455
340282366920938463463374607431768211455
340282366920938463463374607431768211455
340282366920938463463374607431768211455
340282366920938463463374607431768211455
```

wenyongh, 2023-10-31:
Hi, the issue is similar to issue [#2564](https://github.com/bytecodealliance/wasm-micro-runtime/issues/2564#issuecomment-1763972854), for bytecode `f32x4.max 0xffffffff 0xffffffff 0xffffffff 0xffffffff, 0x82cfc799 0xc2f4f1bb 0x23a49ab4 0x01739dab`, the 4 f32 elements of the first vector operand are all NaNs, according to spec, an Nan with with arbitrary content can be returned for f32.max opcode. For wasmtime and wasmer, NaN 0x7fc00000 is returned, for wamr and wasmedge, NaN 0xffffffff  is returned. They are all allowed.

XinyuShe, 2023-11-01:
Thanks for your clarification.