wasm-micro-runtime/wasm-micro-runtime #2688: Related documents of FAST JIT IR?

Rlyown, 2023-10-31:
Hi,
I am very interested in the source code of FAST JIT. But I'm a little unsure about the meaning of the JIT IR instruction. Is there any relevant documentation to help me better understand FAST JIT IR?

For example, for control flow instructions, I guess JIT_OP_BEQ is similar to the ARM BEQ ​​instruction. This might help me understand the behavior of JIT_OP_BEQ. But for the select instruction, I'm not sure if it's similar to the arm CSEL instruction.

I know that reading the code in the codegen pass would allow me to determine what it means, but this is a bit of a stretch for me.

So I'm looking for some enlightening advice to help me understand the code better.

TianlongLiang, 2023-11-02:
Hi, sadly there isn't a formal document for FAST JIT IR. You're right that some instructions, like 'select', don't have corresponding assembly instructions that you can simply one-to-one mapping to. I'd suggest checking if there's a similar instruction in LLVM IR. Generally speaking, FAST JIT IR aligns more closely with LLVM IR in terms of its level of abstraction.

And you can check `core/iwasm/fast-jit/cg/x86-64/jit_codegen_x86_64.cpp` for the code generation implementation of FAST JIT IR that you are interested in to make sure your understanding is correct.

Rlyown, 2023-11-02:
@TianlongLiang Thanks for your advice.   I'll check `LLVM IR` and `codegen` later.