wasm-micro-runtime/wasm-micro-runtime #2710: Compare instruction can't be compiled when meeting inf

unicornt, 2023-11-02:
Hi, I noticed that `fast-jit` will throw `compilation failed` when meeting `inf` in `compare instruction` like `f64.le`.

### Test Case
```wasm
(module
  (type (;0;) (func (param i32)))
  (type (;1;) (func))
  (type (;2;) (func (result i32)))
  (type (;3;) (func (param i32) (result i32)))
  (import "wasi_snapshot_preview1" "proc_exit" (func (;0;) (type 0)))
  (func (;1;) (type 1)
    i32.const 0
    call 2
    drop
    nop)
  (func (;2;) (type 3) (param i32) (result i32)
    (local f64)
    f64.const inf (;=inf;)
    f64.const 1.05 (;=inf;)
    f64.le)
  (table (;0;) 2 2 funcref)
  (memory (;0;) 256 256)
  (global (;0;) (mut i32) (i32.const 66576))
  (global (;1;) (mut i32) (i32.const 66576))
  (export "memory" (memory 0))
  (export "__indirect_function_table" (table 0))
  (export "_start" (func 1))
  (elem (;0;) (i32.const 1) func 1)
  (data (;0;) (i32.const 1024) "5\89v"))
```

[case.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13234591/report16.zip)

### Result
[image]
[image]

By the way, I think maybe `wamr` should **not** return `code 0` when meeting `compilation failed` exception.

wenyongh, 2023-11-03:
@unicornt thanks for reporting the issue! Please try #2718.

unicornt, 2023-11-03:
Yes, it really works. Thank you for the reply!

closed 2023-11-03 by unicornt