wasm-micro-runtime/wasm-micro-runtime #2721: AOT compilation encountered performance issues

XinyuShe, 2023-11-05:
[filea11778.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13259656/filea11778_instr_instr.zip)

[image]
[image]

I tested the AOT compilation, but found that some of the files timed out, taking more than a minute to compile, while the other runtimes finished compiling very quickly.

lum1n0us, 2023-11-07:
😮‍💨 yes, we've noticed that issue for a long time. The whole process of AOT compilation is very similar with a native binary compilation. Linking and optimization are top2 time consumers. Especially, linking involves I/O processing, and it is sequential for now. We are still looking for a solution to speed up. We are also open to any suggestion.

XinyuShe, 2023-11-07:
Could you mark this issue as a performance bug?