wasm-micro-runtime/wasm-micro-runtime #2722: Whether LLVM JIT mode supports invoking the export function with results or not

XinyuShe, 2023-11-05:
[image]
[image]
I have a binary wasm file with a return value. When I execute it using 'iwasm filea2.wasm --heap-size=0 -f main --llvm-jit', the result is as follows:
```bash
[06:52:18:872 - 7F5B183DB740]: WASM execute application failed: invalid main function type.

Exception: invalid function type of main function
```

lum1n0us, 2023-11-06:
Please try `iwasm --heap-size=0 -f main --llvm-jit file2.wasm`

XinyuShe, 2023-11-06:
I have tested this , but the result is :
[image]

XinyuShe, 2023-11-07:
I have tried placing file2.wasm in every possible location, but each attempt resulted in an issue. @lum1n0us

lum1n0us, 2023-11-07:
Guess you are using a self-building *iwasm* and it is compiled with interpreter only option. In that case, `--llvm-jit` is not necessary. 

The root cause of the original problem, *invalid function type of main function*, is putting iwasm command options, `--heap-size=0 -f main --llvm-jit` in your case, behind the Wasm file path, `filea2.wasm` in your case, which tells that those options are for the Wasm file.

To be exactly,

`iwasm filea2.wasm --heap-size=0 -f main --llvm-jit` means running file2.wasm. `--heap-size=0 -f main --llvm-jit` are options for *filea2.wasm*.

It should be

`iwasm --heap-size=0 -f main --llvm-jit filea2.wasm`. `--heap-size=0 -f main --llvm-jit` are options  for `iwasm`.