wasm-micro-runtime/wasm-micro-runtime #2726: Heap overflow in wasm_loader_prepare_bytecode

goldds96, 2023-11-06:
## Environments

**OS** : Ubuntu 18.04 5.4.0-150-generic
**Commit** : 52db362b897221c2a438197a0f2e4a9a300becd4
**Version** : iwasm 1.2.3

## Vulnerability Description

**Affected Tool** : iwasm (linux build)
**Affected Version** : = 1.2.3
**Impact** : Heap out of bounds read (size of 1)

* There is Heap out of bound read of size 1 in wasm_loader_prepare_bytecode function in /core/iwasm/interpreter/wasm_loader.c:7141

## PoC
**1. Input File**

[heap-overflow-poc.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13270590/heap-overflow-poc.zip)

**2. Reproduce**

```shell
$ ~/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm poc
```

**3. ASAN**

```shell
$ ~/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm poc

==14712==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6030000000ea at pc 0x564acc92478f bp 0x7ffe13134890 sp 0x7ffe13134880
READ of size 1 at 0x6030000000ea thread T0
    #0 0x564acc92478e in wasm_loader_prepare_bytecode /home/dskim/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:7141
    #1 0x564acc92478e in load_from_sections /home/dskim/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:3561
    #2 0x564acc929266 in load /home/dskim/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:3921
    #3 0x564acc929266 in wasm_loader_load /home/dskim/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:4088
    #4 0x564acc80e8cf in wasm_runtime_load /home/dskim/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:1228
    #5 0x564acc7fc090 in main /home/dskim/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:931
    #6 0x7f0c717bfc86 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21c86)
    #7 0x564acc800529 in _start (/home/dskim/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm+0x24529)

0x6030000000ea is located 0 bytes to the right of 26-byte region [0x6030000000d0,0x6030000000ea)
allocated by thread T0 here:
    #0 0x7f0c7242eb40 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb40)
    #1 0x564acc801729 in bh_read_file_to_buffer /home/dskim/wasm-micro-runtime/core/shared/utils/uncommon/bh_read_file.c:96

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/dskim/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:7141 in wasm_loader_prepare_bytecode
Shadow bytes around the buggy address:
  0x0c067fff7fc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c067fff7fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c067fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c067fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c067fff8000: fa fa 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00
=>0x0c067fff8010: 00 00 fa fa fd fd fd fd fa fa 00 00 00[02]fa fa
  0x0c067fff8020: 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00 00 00
  0x0c067fff8030: fa fa 00 00 00 00 fa fa fa fa fa fa fa fa fa fa
  0x0c067fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c067fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c067fff8060: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==14712==ABORTING
```

wenyongh, 2023-11-08:
@goldds96 thanks for reporting the issue! Please try PR #2734.

wenyongh, 2024-01-05:
Close this issue as it was fixed.

closed 2024-01-05 by wenyongh