wasm-micro-runtime/wasm-micro-runtime #2743: Using on ESP32 with Rust

AlixANNERAUD, 2023-11-09:
Hello,
I'have tried with WAMR In Rust on linux platform and it worked well (I forked and updated the rust bindgen `wamr-sys` which is no longer maintained : [https://github.com/AlixANNERAUD/wamr-sys.git](https://github.com/AlixANNERAUD/wamr-sys.git)). Now, I would like to use it on an ESP32 using Rust. I've seen that you already support the ESP-IDF which is great. But how it with Rust ? I've read your [warm porting guide](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/port_wamr.md) but I'm quite lost and don't know where to start.

lum1n0us, 2023-11-11:
We are working on an official Rust SDK. You will see details [here](https://github.com/bytecodealliance/wasm-micro-runtime/issues/2684).

AlixANNERAUD, 2023-11-16:
I've found a solution to the problem that's not too bad (thanks to [ivmarkov](https://github.com/ivmarkov)). I use WAMR as an [ESP-IDF component](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#extra-esp-idf-components) by adding the followings options to the `cargo.toml` file:
```yaml
[[package.metadata.esp-idf-sys.extra_components]]
bindings_header = "<path to wamr>/core/iwasm/include/wasm_export"
component_dirs = ["<path to wamr>/build-scripts/esp-idf"]
name = "wamr"
```
The esp-idf-sys crate will then compile WAMR and provide the (unsafe) bindings needed to use WAMR in Rust. However, this solution is far from being the best, and I think it's possible to do better.

closed 2024-10-12 by AlixANNERAUD