wasm-micro-runtime/wasm-micro-runtime #2764: passive/declarative segments should be dropped on instantiation

yamt, 2023-11-14:
it doesn't seem implement correctly, at least for the classic interpreter.

yamt, 2023-11-14:
i guess i will take a look after fixing https://github.com/bytecodealliance/wasm-micro-runtime/issues/2735

sjamesr, 2024-11-20:
I just came across this while trying to eliminate spec test failures, is this as straightforward as something like:

```
diff --git a/core/iwasm/interpreter/wasm_runtime.c b/core/iwasm/interpreter/wasm_runtime.c
--- a/core/iwasm/interpreter/wasm_runtime.c
+++ b/core/iwasm/interpreter/wasm_runtime.c
@@ -2467,8 +2467,9 @@ wasm_instantiate(WASMModule *module, WAS
             goto fail;
         }
         for (i = 0; i < module->table_seg_count; i++) {
-            if (wasm_elem_is_active(module->table_segments[i].mode))
-                bh_bitmap_set_bit(module_inst->e->common.elem_dropped, i);
+          if (wasm_elem_is_active(module->table_segments[i].mode) ||
+              wasm_elem_is_declarative(module->table_segments[i].mode))
+            bh_bitmap_set_bit(module_inst->e->common.elem_dropped, i);
         }
     }
 #endif

```