wasm-micro-runtime/wasm-micro-runtime #2782: Make Zephyr port work as a Zephyr module

hasheddan, 2023-11-16:
There is currently a [Zephyr module file](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/zephyr/module.yml) in this repository, but it cannot be used because it references `cmake-ext` and `kconfig-ext`, despite `CMakeLists.txt` and `Kconfig` not being present for WAMR in [Zephyr upstream modules](https://github.com/zephyrproject-rtos/zephyr/tree/main/modules). There was discussion about including upstream in https://github.com/zephyrproject-rtos/zephyr/issues/21329, but it appears that the decision for now is for it to live separately. With that being the case, we should make it easy for Zephyr users to pull WAMR in as an external module. This primarily just consists of adding an appropriate `CMakeLists.txt` and `Kconfig` in the WAMR repository, then pointing to them via the `zephyr/module.yml` file.

I'd be happy to make these updates -- @wenyongh what do you think?

wenyongh, 2023-11-17:
@hasheddan Yes, that's great! Shall we add the `CMakeLists.txt` and `Kconfig` under `<wamr_root>/zephry` folder? There is a `CMakeLists.txt` under `<wamr_root>` already.

hasheddan, 2023-11-17:
@wenyongh either there or https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/shared/platform/zephyr is great with me -- do you have a preference? I'll get started adding it!

wenyongh, 2023-11-18:
@hasheddan had better put them under `<wamr_root>/zephry`, `core/shared/platform/zephyr` may be too deep and not easy to maintain.

hasheddan, 2023-11-18:
@wenyongh will do, thanks!

nguterresn, 2025-02-22:
@hasheddan hi!
I stumbled upon the same issue as you — when including the WAMR in the same west.yml as the other modules, the Zephyr looks up for the Kconfig but it doesn't exist.
Were you able to fix this?

FabienCesari, 2025-08-05:
I figured out how to get it to build without the Kconfig and CMakeLists.txt - You should not import WAMR as a zephyr module from the yaml file but instead clone the repo in the module folder. You then will have to modify your project CMakeLists.txt to add the path to the WAMR source and relevant config flags.