wasm-micro-runtime/wasm-micro-runtime #2787: iwasm can `sock_shutdown` non-socket file descriptor

yagehu, 2023-11-19:
`sock_shutdown` on a non-socket file descriptor should fail with `57 notsock`. However, iwasm reports no error.

https://github.com/WebAssembly/WASI/blob/2980bb39e1d2a4a2adae4748908cb4325cd41a26/legacy/preview1/docs.md#-sock_shutdownfd-fd-how-sdflags---result-errno

Short WASI program to repro:

```wat
(module
  (type (func))
  (type (func (param i32 i32) (result i32)))
  (type (func (param i32)))
  (import "wasi_snapshot_preview1" "sock_shutdown" (func $sock_shutdown (type 1)))
  (import "wasi_snapshot_preview1" "proc_exit" (func $proc_exit (type 2)))
  (func $_start
    (call $sock_shutdown
      (i32.const 3)
      (i32.const 3)
    )
    call $proc_exit
  )
  (memory 2)
  (export "memory" (memory 0))
  (export "_start" (func $_start))
)
```

This should exit with status code 57. But running this with iwasm reports 0.

wenyongh, 2023-11-20:
@loganek could you help check the issue? Thanks a lot.

loganek, 2023-11-20:
Hi, there's indeed a problem with the implementation. I've opened a PR to fix it on POSIX platforms for now just to quickly unblock you (I assume your platform is POSIX? let me know otherwise); I'd suggest keeping the issue open until we address the implementation for all the other platforms though.

https://github.com/bytecodealliance/wasm-micro-runtime/pull/2801

yagehu, 2023-11-24:
Yes. I'm on POSIX. But I'll leave it up to you folks to close when appropriate. Thanks for the quick fix!

loganek, 2023-11-24:
I've ended up implementing this for all the platforms. @yagehu could you please confirm the latest WAMR version from git solves your problem? If so, please resolve the ticket.

yagehu, 2023-11-25:
Confirmed on amd64 Linux. Closing. Thanks @loganek

closed 2023-11-25 by yagehu