wasm-micro-runtime/wasm-micro-runtime #2804: AOT data race with pigz.wasm test app

eloparco, 2023-11-20:
Hello, I'm running the pigz app from https://github.com/bytecodealliance/wasm-parallel-gzip on Ubuntu and running into some TSAN warnings:
```bash
./iwasm -v=5 --dir=wasm-apps wasm-apps/pigz.aot -p 3 wasm-apps/test.txt
[00:44:04:764 - 7F7C90D8A380]: Load module success.

[00:44:04:821 - 7F7C90D8A380]: Memory instantiate:
[00:44:04:825 - 7F7C90D8A380]:   page bytes: 65536, init pages: 4, max pages: 1024
[00:44:04:827 - 7F7C90D8A380]:   data offset: 0, stack size: 0
[00:44:04:829 - 7F7C90D8A380]:   heap offset: 262144, heap size: 0

==================
WARNING: ThreadSanitizer: data race (pid=185402)
  Write of size 8 at 0x7f7a8dda6360 by thread T3:
    #0 memmove ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:810 (libtsan.so.0+0x6225e)
    #1 memmove ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:808 (libtsan.so.0+0x6225e)
    #2 aot_memmove wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:2315 (iwasm+0x34f71)
    #3 <null> <null> (0x000041271755)
    #4 invoke_native_with_hw_bound_check wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1445 (iwasm+0x32715)
    #5 aot_call_function wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1654 (iwasm+0x32f71)
    #6 wasm_runtime_call_wasm wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2037 (iwasm+0x13af4)
    #7 thread_start wasm-micro-runtime/core/iwasm/libraries/lib-wasi-threads/lib_wasi_threads_wrapper.c:60 (iwasm+0x79a5c)
    #8 thread_manager_start_routine wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c:602 (iwasm+0x36d2d)
    #9 os_thread_wrapper wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:46 (iwasm+0xcb0d)

  Previous read of size 8 at 0x7f7a8dda6360 by thread T1:
    #0 writev ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:1203 (libtsan.so.0+0x4ce32)
    #1 os_writev wasm-micro-runtime/core/shared/platform/common/posix/posix_file.c:578 (iwasm+0x391e2)
    #2 blocking_op_writev wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/blocking_op.c:56 (iwasm+0x86171)
    #3 wasmtime_ssp_fd_write wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:1120 (iwasm+0x42b37)
    #4 wasi_fd_write wasm-micro-runtime/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c:669 (iwasm+0x81ebd)
    #5 <null> <null> (0x000041280a02)
    #6 invoke_native_with_hw_bound_check wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1445 (iwasm+0x32715)
    #7 aot_call_function wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1654 (iwasm+0x32f71)
    #8 wasm_runtime_call_wasm wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2037 (iwasm+0x13af4)
    #9 thread_start wasm-micro-runtime/core/iwasm/libraries/lib-wasi-threads/lib_wasi_threads_wrapper.c:60 (iwasm+0x79a5c)
    #10 thread_manager_start_routine wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c:602 (iwasm+0x36d2d)
    #11 os_thread_wrapper wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:46 (iwasm+0xcb0d)

  Thread T3 (tid=185406, running) created by main thread at:
    #0 pthread_create ../../../../src/libsanitizer/tsan/tsan_interceptors_posix.cpp:969 (libtsan.so.0+0x605b8)
    #1 os_thread_create_with_prio wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:86 (iwasm+0xcd1e)
    #2 os_thread_create wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:100 (iwasm+0xcdb8)
    #3 wasm_cluster_create_thread wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c:704 (iwasm+0x37123)
    #4 thread_spawn_wrapper wasm-micro-runtime/core/iwasm/libraries/lib-wasi-threads/lib_wasi_threads_wrapper.c:122 (iwasm+0x79d7a)
    #5 <null> <null> (0x000041281d65)
    #6 invoke_native_with_hw_bound_check wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1445 (iwasm+0x32715)
    #7 aot_call_function wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1654 (iwasm+0x32f71)
    #8 wasm_runtime_call_wasm wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2037 (iwasm+0x13af4)
    #9 execute_main wasm-micro-runtime/core/iwasm/common/wasm_application.c:112 (iwasm+0xf00a)
    #10 wasm_application_execute_main wasm-micro-runtime/core/iwasm/common/wasm_application.c:238 (iwasm+0xf608)
    #11 app_instance_main wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:112 (iwasm+0xa69c)
    #12 main wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:941 (iwasm+0xc177)

  Thread T1 (tid=185404, running) created by main thread at:
    #0 pthread_create ../../../../src/libsanitizer/tsan/tsan_interceptors_posix.cpp:969 (libtsan.so.0+0x605b8)
    #1 os_thread_create_with_prio wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:86 (iwasm+0xcd1e)
    #2 os_thread_create wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:100 (iwasm+0xcdb8)
    #3 wasm_cluster_create_thread wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c:704 (iwasm+0x37123)
    #4 thread_spawn_wrapper wasm-micro-runtime/core/iwasm/libraries/lib-wasi-threads/lib_wasi_threads_wrapper.c:122 (iwasm+0x79d7a)
    #5 <null> <null> (0x000041281d65)
    #6 invoke_native_with_hw_bound_check wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1445 (iwasm+0x32715)
    #7 aot_call_function wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1654 (iwasm+0x32f71)
    #8 wasm_runtime_call_wasm wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2037 (iwasm+0x13af4)
    #9 execute_main wasm-micro-runtime/core/iwasm/common/wasm_application.c:112 (iwasm+0xf00a)
    #10 wasm_application_execute_main wasm-micro-runtime/core/iwasm/common/wasm_application.c:238 (iwasm+0xf608)
    #11 app_instance_main wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:112 (iwasm+0xa69c)
    #12 main wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:941 (iwasm+0xc177)

SUMMARY: ThreadSanitizer: data race wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:2315 in aot_memmove
==================
[... 124 lines cut ...]
  output format: AoT file
Compile success, file wasm-apps/pigz.aot was generated.
$ head -c 10M </dev/urandom >wasm-apps/test.txt  # Generate random file
$ rm -rf wasm-apps/test.txt.gz  # Remove previously compressed random file
$ ninja && ./iwasm -v=5 --dir=wasm-apps wasm-apps/pigz.aot  -p 3 wasm-apps/test.txt
```

How I added the sanitizer:
```bash
git diff
diff --git a/samples/wasi-threads/CMakeLists.txt b/samples/wasi-threads/CMakeLists.txt
index 467f5fd1..135dbd86 100644
--- a/samples/wasi-threads/CMakeLists.txt
+++ b/samples/wasi-threads/CMakeLists.txt
@@ -70,6 +70,9 @@ add_subdirectory(wasm-apps)
 ################ wamr runtime ################
 include (${SHARED_DIR}/utils/uncommon/shared_uncommon.cmake)
 
+set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -fsanitize=thread")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fsanitize=thread")
+
 set (RUNTIME_SOURCE_ALL
     ${CMAKE_CURRENT_LIST_DIR}/../../product-mini/platforms/linux/main.c
     ${UNCOMMON_SHARED_SOURCE}
```

wenyongh, 2023-11-23:
@eloparco I have a try, the `aot_memmove` and `aot_memset` are called from the AOT code when bulk memory feature is enabled, in which the wasm opcode `memory.copy` and `memory.fill` are compiled into calling `aot_memmove` and `aot_memset`. Acquiring the global shared memory lock in `aot_memove` can make the TSAN warning disappear:
```diff
diff --git a/core/iwasm/aot/aot_runtime.c b/core/iwasm/aot/aot_runtime.c
index 076574cb..6b8e170a 100644
--- a/core/iwasm/aot/aot_runtime.c
+++ b/core/iwasm/aot/aot_runtime.c
@@ -2312,13 +2312,19 @@ aot_check_app_addr_and_convert(AOTModuleInstance *module_inst, bool is_str,
 void *
 aot_memmove(void *dest, const void *src, size_t n)
 {
-    return memmove(dest, src, n);
+    os_mutex_lock(&g_shared_memory_lock);
+    void *ret = memmove(dest, src, n);
+    os_mutex_unlock(&g_shared_memory_lock);
+    return ret;
 }

 void *
 aot_memset(void *s, int c, size_t n)
 {
-    return memset(s, c, n);
+    os_mutex_lock(&g_shared_memory_lock);
+    void *ret = memset(s, c, n);
+    os_mutex_unlock(&g_shared_memory_lock);
+    return ret;
 }

 double
```
I think we can enhance the implementation to call runtime APIs to acquire/release shared memory in AOT code before calling aot_memmove/aot_memset, e.g. call aot_shared_mem_lock/unlock. But not sure whether it is enough, should we also acquire the lock in the libc-wasi native, e.g. `wasi_fd_read`/`wasi_fd_write`/`wasi_fd_pread`/`wasi_fd_pwrite`, or even each time calling the native API?

loganek, 2023-11-23:
I think I saw that in wasmtime; every time the wasi function is called, it locks the memory mutex, to make sure the linear memory is protected. I don't think we need to do this for every wasi function, but at least the ones that use linear memory should have the lock acquired, otherwise there's a chance for concurrent unprotected memory access.

wenyongh, 2023-11-23:
Yes, we had better acquire the lock only for the native functions that use linear memory, but it is not easy to know whether a wasi function uses linear memory or not in compilation time and difficult to emit lock/unlock shared memory IR before/after calling the native function. I think we had better check the implementation in libc_wasi_wrapper.c, and add lock one by one if needed, how do you think?

loganek, 2023-11-23:
Completely agree, I don't think there's an automatic way to find all the WASI functions that access linear memory (the call of `validate_native_addr` gives us a hint, but what if we forgot to add the validation or the memory is allocated using `wasm_runtime_malloc` somewhere in the implementation). On the other hand, there's (only?) about 100 functions to check and for most of them it's very clear whether the lock is necessary or not.

I'll speak to my team today about that and let you know what'd the plan be. 

About the memmove/memset functions - are you planning to open a PR with your fix?

wenyongh, 2023-11-23:
> Completely agree, I don't think there's an automatic way to find all the WASI functions that access linear memory (the call of `validate_native_addr` gives us a hint, but what if we forgot to add [...]

I have submitted PR #2815 to fix this issue, currently it should work for the pigz.wasm with AOT mode when tsan is enabled. I didn't add lock for the wasi function, could your team help do that after this PR is merged? Thanks.

wenyongh, 2023-11-23:
> @wenyongh just one more thought on that; users can expose custom native functions to WASM; don't you think we should in that case expose a public API for them to acquire the lock in case the native function accesses linear memory? I think that's no different to WASI functions.

Yes, how about `void wasm_runtime_shared_mem_lock(wasm_module_inst_t module_inst)` and `void wasm_runtime_shared_mem_unlock(wasm_module_inst_t module_inst)`?

loganek, 2023-11-23:
> I didn't add lock for the wasi function, could your team help do that after this PR is merged?

Quite likely, but I'll confirm that later today.

> Yes, how about void wasm_runtime_shared_mem_lock(wasm_module_inst_t module_inst) and void wasm_runtime_shared_mem_unlock(wasm_module_inst_t module_inst)

Yes, that's more or less what I've had in mind.

g0djan, 2023-11-23:
> I didn't add lock for the wasi function

do you mean wasi functions that access linear memory?

wenyongh, 2023-11-24:
> > I didn't add lock for the wasi function
> 
> do you mean wasi functions that access linear memory?

Yes.

loganek, 2023-11-24:
@wenyongh we've discussed the case with the team and came to a conclusion that it should rather be user's responsibility to add or remove a lock in the application itself - there's no need to lock the entire memory, but rather only a region that's being accessed by the specific function - and that can be achieved only by the application. When it comes to allocating the memory; I'd expect user in multithreaded application to use thread-safe malloc, so that shouldn't be an issue too.

Having said that, this unfortunately complicates TSAN validation a bit - from TSAN's perspective, there are no 'regions' in the linear memory and the whole memory is treated as one blob; I'm not sure what'd the best path forward to it would be - one way is that we'll have a special TSAN build in the CI that adds a lock for all the WASI functions, although I'd like to know others' opinion on that as there might be a better solution.

wenyongh, 2023-11-24:
@loganek Yes, sounds good to let user add or remove lock by himself, and we can provide the API to acquire and release lock. For the TSAN, I have no good idea also. If to add a lock for all the WASI functions, one way I can find is to call runtime API to acquire the lock in AOT code before calling native functions, but it may need to add an option for wamrc.

BTW, could you help review PR #2815? Should we add lock for aot_memmove and aot_memset?

loganek, 2023-11-24:
> BTW, could you help review PR https://github.com/bytecodealliance/wasm-micro-runtime/pull/2815? Should we add lock for aot_memmove and aot_memset?

I think the problem with `aot_memmove` and `aot_memset` is similar to WASI functions - most likely the lock is already acquired in WASM but TSAN doesn't know about that. If that's the case, I don't think we need the PR, users should lock the memory instead in WASM.
@eloparco - could you confirm that?

>  we can provide the API to acquire and release lock.

Hmm, not sure if that's actually needed - user can create a global lock in WASM and use that instead.

wenyongh, 2023-11-24:
> I think the problem with `aot_memmove` and `aot_memset` is similar to WASI functions - most likely the lock is already acquired in WASM but TSAN doesn't know about that. If that's the case, I don't [...]

OK, let's wait for @eloparco's confirmation, please let me know whether to add lock for aot_memmove, or just keep the API exported.

eloparco, 2023-11-27:
Agree that locking the entire shared memory is overkill.
Trying to understand if the TSAN problem is a false alarm or if it's coming from the WASM app (wasm-parallel-gzip) not using locks properly when dealing with shared variables. But in the latter case I would expect a data race been detected in WASM and not only AOT.

eloparco, 2023-11-27:
How can I understand what function is calling those `aot_memmove` and `aot_memset` and what function is trying to read from those locations? To understand if the problem is coming from the runtime, the wasm code or it's a false alarm from TSAN.

eloparco, 2023-11-27:
I added logs for every time a mutex is acquired and released in the wasm-parallel-gzip app, but I didn't discover anything useful
```bash
+ ./iwasm -v=5 --dir=wasm-apps wasm-apps/pigz.aot -p 3 wasm-apps/test.txt
pigz acquiring lock 0x28024
pigz releasing lock 0x28024
pigz acquiring lock 0x38cd0
pigz releasing lock 0x38cd0
pigz acquiring lock 0x38c80
pigz acquiring lock 0x38cd0
pigz releasing lock 0x38cd0
pigz acquiring lock 0x48ef0
pigz acquiring lock 0x28024
pigz releasing lock 0x28024
pigz acquiring lock 0x38c30
pigz acquiring lock 0x38cd0
pigz releasing lock 0x38cd0
pigz acquiring lock 0x38c30
pigz acquiring lock 0x38d20
pigz releasing lock 0x38d20
pigz acquiring lock 0x68ff0
pigz acquiring lock 0x28024
pigz releasing lock 0x28024
pigz acquiring lock 0x38c30
pigz acquiring lock 0x38cd0
pigz releasing lock 0x38cd0
pigz acquiring lock 0x38c30
pigz acquiring lock 0xda8b0
pigz acquiring lock 0x28024
pigz acquiring lock 0x48ef0
pigz acquiring lock 0x38d20
pigz releasing lock 0x38d20
pigz releasing lock 0x28024
pigz acquiring lock 0x38c30
pigz acquiring lock 0x38cd0
pigz releasing lock 0x38cd0
pigz acquiring lock 0x38c30
pigz acquiring lock 0x134af0
pigz acquiring lock 0x38c30
pigz acquiring lock 0x38cd0
pigz releasing lock 0x38cd0
pigz acquiring lock ==================
WARNING: ThreadSanitizer: data race (pid=864277)
  Write of size 8 at 0x7fb3b6028718 by main thread:
    #0 memmove ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:810 (libtsan.so.0+0x6225e)
    #1 memmove ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:808 (libtsan.so.0+0x6225e)
    #2 aot_memmove /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:2315 (iwasm+0x34f75)
    #3 <null> <null> (0x000040e35699)
    #4 invoke_native_with_hw_bound_check /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1445 (iwasm+0x32719)
    #5 aot_call_function /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1654 (iwasm+0x32f75)
    #6 wasm_runtime_call_wasm /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2042 (iwasm+0x13afd)
    #7 execute_main /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/common/wasm_application.c:112 (iwasm+0xf013)
    #8 wasm_application_execute_main /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/common/wasm_application.c:238 (iwasm+0xf611)
    #9 app_instance_main /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:113 (iwasm+0xa6a5)
    #10 main /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:947 (iwasm+0xc180)

  Previous read of size 8 at 0x7fb3b6028718 by thread T4:
    #0 writev ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:1203 (libtsan.so.0+0x4ce32)
    #1 os_writev /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/shared/platform/common/posix/posix_file.c:578 (iwasm+0x391e6)
    #2 blocking_op_writev /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/blocking_op.c:56 (iwasm+0x86173)
    #3 wasmtime_ssp_fd_write /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:1120 (iwasm+0x42b3b)
    #4 wasi_fd_write /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c:669 (iwasm+0x81ebf)
    #5 <null> <null> (0x000040e36682)
    #6 invoke_native_with_hw_bound_check /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1445 (iwasm+0x32719)
    #7 aot_call_function /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1654 (iwasm+0x32f75)
    #8 wasm_runtime_call_wasm /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2042 (iwasm+0x13afd)
    #9 thread_start /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/libraries/lib-wasi-threads/lib_wasi_threads_wrapper.c:60 (iwasm+0x79a3d)
    #10 thread_manager_start_routine /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c:602 (iwasm+0x36d31)
    #11 os_thread_wrapper /home/ANT.AMAZON.COM/eloparco/dev/wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:46 (iwasm+0xcb16)

  Thread T4 (tid=864282, running) created by main thread at:
[... 2022 lines cut ...]
pigz acquiring lock 0x38c30
pigz acquiring lock 0x28024
pigz releasing lock 0x38c30
pigz acquiring lock 0x28024
pigz releasing lock 0x38c30
pigz releasing lock 0x38c30
pigz acquiring lock 0x28024
pigz acquiring lock 0x28024
pigz releasing lock 0x28024
pigz acquiring lock 0x38dc0
pigz releasing lock 0x38dc0
pigz acquiring lock 0x38d70
pigz releasing lock 0x38d70
pigz acquiring lock 0x38d20
pigz releasing lock 0x38d20
pigz acquiring lock 0x38cd0
pigz releasing lock 0x38cd0
ThreadSanitizer: reported 10 warnings
[04:48:29:745 - 7FB5B83DE380]: Load module success.

[04:48:29:820 - 7FB5B83DE380]: Memory instantiate:
[04:48:29:823 - 7FB5B83DE380]:   page bytes: 65536, init pages: 4, max pages: 1024
[04:48:29:825 - 7FB5B83DE380]:   data offset: 0, stack size: 0
[04:48:29:827 - 7FB5B83DE380]:   heap offset: 262144, heap size: 0
```

eloparco, 2023-11-27:
The print was added here https://github.com/madler/pigz/blob/master/yarn.c#L136
```c
void possess_(lock *bolt, char const *file, long line) {
  printf("pigz acquiring lock %p\n", &bolt->mutex); // <------ 
  int ret = pthread_mutex_lock(&(bolt->mutex));
  if (ret)
    fail(ret, file, line, "mutex_lock");
}

void release_(lock *bolt, char const *file, long line) {
  printf("pigz releasing lock %p\n", &bolt->mutex); // <------ 
  int ret = pthread_mutex_unlock(&(bolt->mutex));
  if (ret)
    fail(ret, file, line, "mutex_unlock");
}
```

eloparco, 2023-11-27:
>Actually, I was running the .wasm file (no AOT) with the sanitizer for the latest main and getting another race:

Taking a closer look, those data races match the ones in AOT since I was able to patch them locally by adding some locks around `WASM_OP_MEMORY_FILL` and `WASM_OP_MEMORY_COPY`.

eloparco, 2023-11-28:
I was taking a second look, basically what's causing the race conditions is the access to `memory->memory_data_size`.

We can't just use a mutex to access it because when the HW bound check is disabled, `memory->memory_data_size` is accessed for every memory operation from [CHECK_MEMORY_OVERFLOW](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/interpreter/wasm_interp_classic.c#L44) and the performance impact is huge. So we probably want to use atomics for it.

eloparco, 2023-11-29:
I couldn't re-open the issue. 

The fact that it's only happening in AOT makes me suspicious that maybe it's a bug in AOT mode and not a problem with the wasm application not protecting access to memory. But I didn't find a way to verify it.

wenyongh, 2023-11-30:
@eloparco this issue was closed automatically when PR #2834 was merged. I re-open it now.

eloparco, 2023-12-04:
Any suggestions on how to debug the problem? Couldn't find anything useful.

wenyongh, 2023-12-05:
> Any suggestions on how to debug the problem? Couldn't find anything useful.

@eloparco any issue on the AOT?

eloparco, 2023-12-05:
Yes, still looking into the TSAN problem with AOT, not sure how to debug it. 

I added some prints in the pigz app to understand which app functions call `memmove` and `wasi_fd_write` (and see if they're acquiring the lock properly), but then the sanitizer error disappeared.

wenyongh, 2023-12-05:
The aot_memmove should be called by memory.copy opcode, maybe you can check which function has the opcode? It is introduced by the bulk memory feature.

BTW, since most of the blocking issues have been resolved, I plan to do some tests and create the new release if things run well. Any concern from you?

eloparco, 2023-12-05:
> BTW, since most of the blocking issues have been resolved, I plan to do some tests and create the new release if things run well. Any concern from you?

Yeah, let's proceed with the release. We can release 1.3.1 in case, once we fix this TSAN issue.

eloparco, 2023-12-11:
I tried native pigz (without compiling to wasm) [with tsan](https://github.com/madler/pigz/blob/master/Makefile#L4-L5) and there are no races detected there.

If I compile to AOT and run the pigz app on my M1 macbook, no TSAN errors either.

g0djan, 2024-01-31:
Hi @wenyongh
I created a smaller sample to reproduce the problem

```
use std::sync::mpsc::channel;

fn main() {
    let (task_sender, task_receiver) = channel();

    std::thread::spawn(move || {
        while let Ok(task) = task_receiver.recv() {
            let mut newStr = task;
            newStr += "a";
        }
    });

    for i in 1..300u64 {
        let task = i.to_string();
        let _ = task_sender.send(task);
    }
}
```

I build it using next commands

```
cargo build --target wasm32-wasi-preview1-threads

~/github/wasm-micro-runtime/wamr-compiler/build/wamrc
--enable-multi-thread \
    --disable-simd \
    --target=x86_64 \
    --target-abi=gnu \
    --size-level=1 \
-o test.aot target/wasm32-wasi-preview1-threads/debug/test.wasm

```

I built WAMR following readme from `samples/wasi-threads/`
Then I ran the executable, if I replace String with int types it doesn't report anything.

`~/github/wasm-micro-runtime/samples/wasi-threads/build/iwasm /path/to/test.aot`

g0djan, 2024-02-02:
I made another sample in C that shows the data race warning, but haven't broken it to syscalls/.wat instructions yet

Any advice would be welcome

```
#include <stdio.h>
#include <string.h>
#include <pthread.h>
#include <assert.h>

pthread_mutex_t mut = PTHREAD_MUTEX_INITIALIZER;

void* func(void* arg) {
    char dest[] = "b";

    for (int i = 0; i < 300; i++) {
        pthread_mutex_lock(&mut);
        printf("%s\n", dest);
        pthread_mutex_unlock(&mut);
    }
    
    

    return NULL;
}

int main() {
    char dest[] = "a";
    assert(pthread_mutex_init(&mut, NULL) == 0 && "Failed to init mutex");
    pthread_t worker;
    pthread_create(&worker, NULL, func, NULL);
    for (int i = 0; i < 300; ++i) {
        pthread_mutex_lock(&mut);
        printf("%s\n", dest);
        pthread_mutex_unlock(&mut);
    }
    pthread_join(worker, NULL);
    assert(pthread_mutex_destroy(&mut) == 0 && "Failed to destroy mutex");
    return 0;
}
```

```
WARNING: ThreadSanitizer: data race (pid=2547016)
  Write of size 1 at 0x7fc3da400ff8 by main thread:
    #0 memmove ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:810 (libtsan.so.2+0x62e94)
    #1 memmove ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:808 (libtsan.so.2+0x62e94)
    #2 memmove /usr/include/x86_64-linux-gnu/bits/string_fortified.h:36 (iwasm+0x25dd0)
    #3 aot_memmove /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:2317 (iwasm+0x25dd0)
    #4 <null> <null> (0x000040771157)
    #5 invoke_native_with_hw_bound_check /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1432 (iwasm+0x21d1f)
    #6 aot_call_function /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1656 (iwasm+0x222c3)
    #7 wasm_runtime_call_wasm /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2088 (iwasm+0x112e2)
    #8 execute_main /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/common/wasm_application.c:112 (iwasm+0xe2eb)
    #9 wasm_application_execute_main /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/common/wasm_application.c:238 (iwasm+0xe2eb)
    #10 app_instance_main /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:115 (iwasm+0xb6ed)
    #11 main /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:955 (iwasm+0xb6ed)

  Previous read of size 1 at 0x7fc3da400ff8 by thread T1:
    #0 writev ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:1200 (libtsan.so.2+0x52746)
    #1 os_writev /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/shared/platform/common/posix/posix_file.c:580 (iwasm+0x28bff)
    #2 blocking_op_writev /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/blocking_op.c:56 (iwasm+0x6775f)
    #3 wasmtime_ssp_fd_write /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:1120 (iwasm+0x2f8bd)
    #4 wasi_fd_write /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c:669 (iwasm+0x6587e)
    #5 <null> <null> (0x000040771a4d)
    #6 invoke_native_with_hw_bound_check /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1432 (iwasm+0x21d1f)
    #7 aot_call_function /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1656 (iwasm+0x222c3)
    #8 wasm_runtime_call_wasm /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:2088 (iwasm+0x112e2)
    #9 thread_start /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/libraries/lib-wasi-threads/lib_wasi_threads_wrapper.c:60 (iwasm+0x5de76)
    #10 thread_manager_start_routine /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c:640 (iwasm+0x26cc4)
    #11 os_thread_wrapper /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:46 (iwasm+0xdc8e)

  Thread T1 (tid=2547018, running) created by main thread at:
    #0 pthread_create ../../../../src/libsanitizer/tsan/tsan_interceptors_posix.cpp:1001 (libtsan.so.2+0x63a59)
    #1 os_thread_create_with_prio /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:86 (iwasm+0xcf91)
    #2 os_thread_create /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:100 (iwasm+0xd057)
    #3 wasm_cluster_create_thread /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c:743 (iwasm+0x27107)
    #4 thread_spawn_wrapper /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/libraries/lib-wasi-threads/lib_wasi_threads_wrapper.c:122 (iwasm+0x5dcf5)
    #5 <null> <null> (0x00004076fab2)
    #6 invoke_native_with_hw_bound_check /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1432 (iwasm+0x21d1f)
    #7 aot_call_function /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c:1656 (iwasm+0x222c3)
[... cut ...]
    #10 wasm_application_execute_main /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/core/iwasm/common/wasm_application.c:238 (iwasm+0xe2eb)
    #11 app_instance_main /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:115 (iwasm+0xb6ed)
    #12 main /home/ANT.AMAZON.COM/godjan/github/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:955 (iwasm+0xb6ed)

SUMMARY: ThreadSanitizer: data race /usr/include/x86_64-linux-gnu/bits/string_fortified.h:36 in memmove
```

g0djan, 2024-02-02:
@yamt @wenyongh could you take a look at this small example and maybe give advice on the next steps please?

Reduced sample to make it closer to WASM instructions

My understanding memmove is called by memory.copy instruction https://github.com/bytecodealliance/wasm-micro-runtime/blob/5e9a693bc4eda60ee565ddba54c0c2b01ee67501/core/iwasm/compilation/aot_emit_memory.c#L1077

Which occurs in the corresponding .wat file only twice, inside $__fwritex and $fwrite. I compile without optimisation to make wat more readable
```
/opt/wasi-sdk/bin/clang 
-target wasm32-wasi-threads 
-O0 
-pthread 
-ftls-model=local-exec 
-z stack-size=32768 
-Wl,--export=__heap_base 
-Wl,--export=__data_end 
-Wl,--shared-memory,--max-memory=1966080 
-Wl,--export=wasi_thread_start 
-Wl,--export=malloc         
-Wl,--export=free 
-o datarace.wasm main.c

~/github/wasm-micro-runtime/wamr-compiler/build/wamrc 
--enable-multi-thread     
--disable-simd     
--target=x86_64     
--target-abi=gnu     
--size-level=1 
-o datarace.aot datarace.wasm
```

```
#include <stdio.h>

#define STACK_SIZE 1024

int expected1 = 0;
int expected2 = 0;
int desired = 1;

int th_stop = 0;
int th = 0;

typedef struct {
    void *stack;
} start_args_t;

void
__wasi_thread_start_C(int thread_id, int *start_arg)
{
    start_args_t *base = (start_args_t *)start_arg;
    char dest[] = "b";

    for (int i = 0; i < 300; i++) {

        while (!__atomic_compare_exchange(&th, &expected1, &desired, 0, __ATOMIC_SEQ_CST, __ATOMIC_SEQ_CST)) {
            __builtin_wasm_memory_atomic_wait32(&th, 1, -1);
            expected1 = 0;
        }

        printf("%s\n", dest);

        __atomic_store_n(&th, 0, __ATOMIC_SEQ_CST);
        __builtin_wasm_memory_atomic_notify(&th, 1);
    }

    __atomic_store_n(&th_stop, 1, __ATOMIC_SEQ_CST);
    __builtin_wasm_memory_atomic_notify(&th_stop, 1);
}

int main() {
    start_args_t base;
    char dest[] = "a";

    char stack[STACK_SIZE];
    base.stack = &stack;
    base.stack += STACK_SIZE;

    int thread_id = __wasi_thread_spawn(&base);
    if (thread_id <= 0) {
        return 4;
    }

    for (int i = 0; i < 300; ++i) {

        while (!__atomic_compare_exchange(&th, &expected2, &desired, 0, __ATOMIC_SEQ_CST, __ATOMIC_SEQ_CST)) {
            __builtin_wasm_memory_atomic_wait32(&th, 1, -1);
            expected2 = 0;
        }

        printf("%s\n", dest);
        

        __atomic_store_n(&th, 0, __ATOMIC_SEQ_CST);
        __builtin_wasm_memory_atomic_notify(&th, 1);
    }

    __builtin_wasm_memory_atomic_wait32(&th_stop, 0, -1);
    return 0;
}
```

wenyongh, 2024-02-04:
@g0djan I have a rough test on the samples you provided, for the first example, if I changed `char dest[] = "b";` to `char *dest = "1234567890";`, I found the string can be printed atomically each time no matter I put `dest` in `func` and `main` as local variables or put it as a global variable, I think that means the `pthread_mutex_lock` successfully acquires the lock. If not, the output may be chaotic, for example sub strings of two thread are dumped in a single line, e.g. "1234123456785678". I doubted that it is issue of libc-wasi but not runtime, I remember that libc-wasi has data race issue.

loganek, 2024-02-05:
I just tried the code; I can reproduce the issue, but when I change stdout to stderr, the datarace isn't detected. I'm going to compare the compiled aot code to see the difference, also will check if there's any special handling of stdout in the WAMR itself.

yamt, 2024-02-05:
> @yamt @wenyongh could you take a look at this small example and maybe give advice on the next steps please?

your test code looks broken.
you should use pthread_create to create threads.
the libc internal locking isn't expected to work unless your threads have valid pthread contexts.

g0djan, 2024-02-05:
>your test code looks broken.
you should use pthread_create to create threads.
the libc internal locking isn't expected to work unless your threads have valid pthread contexts.

I did use pthread_create in an earlier version.
Is it broken https://github.com/bytecodealliance/wasm-micro-runtime/issues/2804#issuecomment-1923573048 ?

g0djan, 2024-02-05:
> @g0djan I have a rough test on the samples you provided, for the first example, if I changed char dest[] = "b"; to char *dest = "1234567890";, I found the string can be printed atomically each time [...]

This make sense, output doesn't get mixed for me either.

The TSAN warning however reported to be between [__wasi_fd_write](https://github.com/WebAssembly/wasi-libc/blob/main/libc-bottom-half/cloudlibc/src/libc/sys/uio/writev.c#L33) and [memcpy](https://github.com/WebAssembly/wasi-libc/blob/a6489a85633ae36dc1d3ec93f85d27ca242feed0/libc-top-half/musl/src/stdio/fwrite.c#L23)

Also output of TSAN shows that no mutexes are locked at all at the moment of data race

eloparco, 2024-02-14:
When running https://github.com/bytecodealliance/wasm-micro-runtime/issues/2804#issuecomment-1923573048, 
the memcpy causing the race condition seems to be coming from inside the fwrite https://github.com/WebAssembly/wasi-libc/blob/212296e4fa1c59f7651b1aefb6a399ed9138d97a/libc-top-half/musl/src/stdio/fwrite.c#L23. 

So both the write and memcpy are coming from the `printf` in the code. Can't understand why that happens though since in the user code there's a lock (`pthread_mutex_lock`). Unless there's an issue with TSAN itself.

yamt, 2024-02-14:
> > your test code looks broken.
> > you should use pthread_create to create threads.
> > the libc internal locking isn't expected to work unless your threads have valid pthread contexts.
> 
> I did use pthread_create in an earlier version. Is it broken [#2804 (comment)](https://github.com/bytecodealliance/wasm-micro-runtime/issues/2804#issuecomment-1923573048) ?

this version looks ok in the regard.

yamt, 2024-02-14:
the mutex implementation in wasi-libc sometimes uses non-atomic load to fetch the lock word. i guess it's what TSAN is compilaining in case of the interpreter.

in case of aot, the lock operations are compiled into native machine code. while i don't know how TSAN works, i suspect wamrc should instrument them for TSAN. otherwise, i don't know how TSAN can know about the serializations. how do you think?

g0djan, 2024-02-14:
>the mutex implementation in wasi-libc sometimes uses non-atomic load to fetch the lock word. i guess it's what TSAN is compilaining in case of the interpreter.

Yep, I guess you refer to https://github.com/WebAssembly/wasi-libc/issues/413

>i suspect wamrc should instrument them for TSAN
I'm not sure I got what you meant. What "Intstrument" means here?

yamt, 2024-02-14:
> > the mutex implementation in wasi-libc sometimes uses non-atomic load to fetch the lock word. i guess it's what TSAN is compilaining in case of the interpreter.
> 
> Yep, I guess you refer to [WebAssembly/wasi-libc#413](https://github.com/WebAssembly/wasi-libc/issues/413)

yes.

> > i suspect wamrc should instrument them for TSAN
> 
> I'm not sure I got what you meant. What "Intstrument" means here?

i meant, i suspect wamrc needs to know -fsanitize=thread equivalent.

g0djan, 2024-02-14:
Ah got it 👍

eloparco, 2024-02-15:
> i suspect wamrc should instrument them for TSAN

That would require wamrc to detect `pthread_mutex_lock/unlock` (and similar) functions and call (i.e. `LLVMBuildCall`) tsan annotation functions like the ones here https://github.com/llvm-mirror/compiler-rt/blob/master/include/sanitizer/tsan_interface.h, if my understanding is correct.

g0djan, 2024-02-19:
> wamrc reorders instructions and puts memmove out of lock for some reason

I checked this version and  both calls `memmove` and `writev` are wrapped with intended atomic instructions, so compiler doesn't reorder these intructions.
[reduced_wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/14334623/reduced_wat.txt)
[reduced_opt_ll.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/14334624/reduced_opt_ll.txt)

I used .wat and LLVM intermediate representation for this sample https://github.com/bytecodealliance/wasm-micro-runtime/issues/2804#issuecomment-1924384534 to match atomic calls and calls reported by TSAN

Cheatsheet for the reduced_opt_ll.txt file:

```
Loop1
210:  %maddr43 = getelementptr inbounds i8, ptr %mem_base_addr, i64 1312
226:  %5 = cmpxchg volatile ptr %maddr43, i32 %data34.us194, i32 %data39.us193 seq_cst seq_cst, align 4
235:  %call.us = tail call i32 @wasm_runtime_atomic_wait(ptr %3, ptr nonnull %maddr43, i64 1, i64 -1, i8 0)
242:  %7 = cmpxchg volatile ptr %maddr43, i32 0, i32 %data39.us seq_cst seq_cst, align 4
- It’s optimised exit from loop after “expected1 = 0;”
250:  %call89 = tail call i32 @"aot_func#19"(ptr %exec_env, i32 %data83, i32 1, i32 1, i32 %data87)
- Inside two calls to stdout_write and later memmove
252:  store atomic volatile i32 0, ptr %maddr43 seq_cst, align 4
253:  %call115 = tail call i32 @wasm_runtime_atomic_notify(ptr %3, ptr nonnull %maddr43, i32 1)

memmove:
250:  %call89 = tail call i32 @"aot_func#19"(ptr %exec_env, i32 %data83, i32 1, i32 1, i32 %data87)
401:  %call112 = tail call i32 @"aot_func#19"(ptr %exec_env, i32 %data106, i32 1, i32 1, i32 %data110)
1665:define i32 @"aot_func#19"(ptr %exec_env, i32 %0, i32 %1, i32 %2, i32 %3) local_unnamed_addr {
-> 1962:  tail call void @llvm.memmove.p0.p0.i32(ptr align 1 %maddr116, ptr align 1 %maddr108, i32 %l6.2, i1 false)

stdout_write also inside aot_func#19:
1850:  %res = tail call i8 @aot_call_indirect(ptr %exec_env, i32 0, i32 %data32, i32 3, ptr nonnull %6)
1920:  %res91 = tail call i8 @aot_call_indirect(ptr %exec_env, i32 0, i32 %data69, i32 3, ptr nonnull %6)

Loop2
362:  %maddr65 = getelementptr inbounds i8, ptr %mem_base_addr, i64 1312
377:  %5 = cmpxchg volatile ptr %maddr65, i32 %data56.us227, i32 %data61.us226 seq_cst seq_cst, align 4
386:  %call98.us = tail call i32 @wasm_runtime_atomic_wait(ptr %1, ptr nonnull %maddr65, i64 1, i64 -1, i8 0)
393:  %7 = cmpxchg volatile ptr %maddr65, i32 0, i32 %data61.us seq_cst seq_cst, align 4
- It’s optimised exit from loop after “expected2 = 0;”
401:  %call112 = tail call i32 @"aot_func#19"(ptr %exec_env, i32 %data106, i32 1, i32 1, i32 %data110)
- Inside two calls to stdout_write and later memmove
403:  store atomic volatile i32 0, ptr %maddr65 seq_cst, align 4
404:  %call138 = tail call i32 @wasm_runtime_atomic_notify(ptr %1, ptr nonnull %maddr65, i32 1)

```

closed 2024-02-23 by eloparco