wasm-micro-runtime/wasm-micro-runtime #2820: Print hard link file number error.

Userzxcvbvnm, 2023-11-24:
# Environment
Windows 10
WAMR-1.2.3
amd64

# Test case
The test case is uploaded as readlink.wasm, it is compiled form readlink.c.
[readlink-wasm.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/13455024/readlink-wasm.txt)
[readlink-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/13455025/readlink-c.txt)

One hard link file (Data/link/hello_hardlink.txt) is pointed to Data/hello.txt, and one soft link file(Data/link/hello_softlink.txt) is pointed to Data/hello.txt.
And the expected hard link number is 1.
And in wasi api (https://github.com/WebAssembly/wasi-libc/blob/main/libc-bottom-half/headers/public/wasi/api.h), nlink means number of hard linke to the file.
    /**
     * Number of hard links to the file.
     */
    __wasi_linkcount_t nlink;

# Reproduce steps 
using command `iwasm --dir=./Data readlink.wasm` .

# Expected result
Successfully get the file fd.
File Type: 4
Number of hard links to the file: 1 
File Size: 33 bytes

# Actual result
Successfully get the file fd.
File Type: 4
Number of hard links to the file: 2 
File Size: 33 bytes

TianlongLiang, 2024-01-04:
Hi, I think it is expected and it is right to be 2. The `Data/hello.txt` itself is a hard link. This is because every file starts with a single hard link, which is the link from its directory entry to the inode representing the file's data. This first link is indeed the file itself. When you create additional hard links to that file, they all refer to the same underlying inode and therefore the same data, and the count of hard links increases accordingly.

You can try this C program, using gcc to compile it to native code and test it out, it should return 2 too.
```C
#include <stdio.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <unistd.h>

int main(int argc, char *argv[]) {
    struct stat fileStat;
    if(argc != 2) {
        fprintf(stderr, "Usage: %s <filename>\n", argv[0]);
        return 1;
    }

    if(stat(argv[1], &fileStat) < 0) {
        perror("Failed to get file stats");
        return 1;
    }

    printf("Number of hard links: %ld\n", (long) fileStat.st_nlink);
    return 0;
}
```

yamt, 2024-06-27:
i agree this is not a bug.