wasm-micro-runtime/wasm-micro-runtime #2828: `fd_pwrite` after setting `fdflags::append` writes from the wrong offset

yagehu, 2023-11-28:
Run the following file compiled with wasi-sdk. It creates a file `tmp/a` and performs this sequence of actions:

1. Write 67 bytes to `tmp/a`.
2. Set the `fdflags::append` flag via `fcntl()`.
3. Write 102 bytes at offset 0 to `tmp/a` with `fd_pwrite`.

```c
#include <fcntl.h>
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>

int main() {
    int f = open("tmp/a", O_CREAT | O_WRONLY);
    if (f == -1) {
        perror("open");
	return 1;
    }

    char buf[67];

    int written = write(f, buf, 67);
    printf("written %d\n", written);

    int ret = fcntl(f, F_SETFL, O_APPEND);
    if (ret == -1) {
	perror("fcntl");
	return 1;
    }

    char buf2[102];

    int written2 = pwrite(f, buf2, 102, 0);
    printf("written %d\n", written2);
}
```

The resulting file should be 102 bytes because `fd_pwrite` should write from offset 0 regardless of the append flag per specification (WASI specifies `fd_pwrite` to be similar to [POSIX `pwritev`](https://pubs.opengroup.org/onlinepubs/9699919799/functions/write.html) which specifies this behavior).

Wasmtime behaves correctly, producing a 102-byte file.

wenyongh, 2023-11-29:
Hi, I tested the result of the gcc native (gcc -O3 -o test main, and then ./test), it also creates a 169-byte file. It is confusing whether we should comply with wasi specification or not, if yes, we may change the behavior of C application. And I found that eventually the libc `pwritev` (or `pwrite`) was called with offset 0, but it is strange that it didn't write the data from offset 0.

yagehu, 2023-11-30:
@wenyongh Definitely a fair point. [`man 2 pwrite`](https://linux.die.net/man/2/pwrite) identifies this as a Linux bug. It was [first raised by Micharl Kerrisk](https://bugzilla.kernel.org/show_bug.cgi?id=43178). However, you can also argue WASI aims to be platform independent. The Wasmtime team has probably fixed it by using cap-std. I leave it up to you WAMR folks to decide if you wontfix it or not. I'm just reporting this deviation from the specification.

yagehu, 2023-12-04:
I did some digging and found that Wasmtime solves it in preview1 by managing the append flag and hiding it from the underlying syscall. It has nothing to do with cap-std, which simply forwards the call to Rust std.

Notably, [here](https://github.com/bytecodealliance/wasmtime/blob/57efd6087933f3e642bc89afd1f838693dfb5b90/crates/wasi-preview1-component-adapter/src/lib.rs#L1618) and [here](https://github.com/bytecodealliance/wasmtime/blob/57efd6087933f3e642bc89afd1f838693dfb5b90/crates/wasi-preview1-component-adapter/src/lib.rs#L2352).

yamt, 2023-12-20:
> I did some digging and found that Wasmtime solves it in preview1 by managing the append flag and hiding it from the underlying syscall. It has nothing to do with cap-std, which simply forwards the [...]

such a "fix" would break an important use case of O_APPEND, which is multiple writers to a single file.

yagehu, 2023-12-20:
> such a "fix" would break an important use case of O_APPEND, which is multiple writers to a single file.

I don't think it would break the use case if implemented correctly considering you don't have multiple processes in WASI.

yamt, 2023-12-20:
> I don't think it would break the use case if implemented correctly considering you don't have multiple processes in WASI.

"multiple writers" can include a host process which is not related to wamr.

yagehu, 2024-04-22:
Since this is a Linux behavior, it would be hard to fix without emulating the WASI fs layer. Closing as wontfix.

closed 2024-04-22 by yagehu