wasm-micro-runtime/wasm-micro-runtime #2831: Unable to call wasm_get_sys_tick_ms from WASM Module

iehfned, 2023-11-28:
I'm trying to make use of the WAMR Api inside my WASM Module. However, whenever I try to run the module, I get the following error during Runtime:
```
I (316) WAMR-P1: Instantiate WASM runtime
[00:00:00:056 - 3FCDCB6C]: warning: failed to link import function (env, wasm_get_sys_tick_ms)
I (326) WAMR-P1: run main() of the application
Exception: failed to call unlinked import function (env, wasm_get_sys_tick_ms)
I (336) WAMR-P1: Deinstantiate WASM runtime
```
When I'm building my WASM Module I'm using `core/app-framework/base/app/timer_api.h` during compilation which is done like this:
```
/opt/wasi-sdk/bin/clang  --target=wasm32        \
        --sysroot=${WAMR_DIR}/wamr-sdk/app/libc-builtin-sysroot   \
        -O3 -z stack-size=4096      \
        -Wl,--initial-memory=65536 \
        -Wl,--allow-undefined-file=${WAMR_DIR}/wamr-sdk/app/libc-builtin-sysroot/share/defined-symbols.txt \
        -Wl,--no-entry -Wl,--export=main                \
        -Wl,--export=__heap_base,--export=__data_end    \
        -Wl,--export=__wasm_call_ctors  \
        -nostdlib \
        -I ${WAMR_DIR}/core/app-framework/base/app/ \
        ping.c -o ping.wasm \
```
From my understanding this error comes from not adding this function to the native symbols via wasm_runtime_register_natives. However this should already be done within WAMR itself, right?
Any ideas why this function might be unable to link?

wenyongh, 2023-11-29:
Hi, `wasm_get_sys_tick_ms` is implemented the base lib of app framework, it isn't compiled into iwasm by default, could you please build iwasm with command below:
```bash
cmake .. -DWAMR_BUILD_APP_FRAMEWORK=1 -DWAMR_BUILD_APP_LIST="WAMR_APP_BUILD_BASE"
```

iehfned, 2023-11-29:
Thanks for your help! That worked!
I was missing the build flags because they are not documented anywhere. Maybe updating this Website https://wamr.gitbook.io/document/wamr-in-practice/tutorial/build-tutorial/build_wamr might be in order.

closed 2023-11-29 by iehfned