wasm-micro-runtime/wasm-micro-runtime #2835: The implementation of the three narrow op codes does not comply with the simd proposal

liutao-liu, 2023-11-29:
The implementation of the three op codes does not comply with the simd proposal : i8x16.narrow_i16x8_u、i16x8.narrow_i32x4_u、i32x4.narrow_i64x2_u.

https://github.com/WebAssembly/relaxed-simd/blob/main/proposals/simd/SIMD.md
[image]

lum1n0us, 2023-11-29:
Would u mind providing the reproducing steps and release version information?

Btw, I run a quick test on above highlight case and got the right answer.

```
webassembly> i8x16.narrow_i16x8_u 0x0\0x0 0xffffffffffffffff\0xffffffffffffffff
<0x0000000000000000 0x0000000000000000>:v128
```

liutao-liu, 2023-12-02:
I am using wamr 1.1.x.  I think you can reproduction the problem i said by wamr aot compiler.

And the specific modifications are as follows:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2850

liutao-liu, 2023-12-04:
i am sorry my modifications  confuse you, i will fix later.

The difference between ```i8x16.narrow_i16x8_s``` and ```i8x16.narrow_i16x8_u``` is that the maximum and minimum values are different when overflow occurs.

lum1n0us, 2023-12-05:
> The difference between i8x16.narrow_i16x8_s and i8x16.narrow_i16x8_u is that the maximum and minimum values are different when overflow occurs.

totally agree with that. But I am not following about the issue raised at the first place. Plus, WAMR v1.2.3 can pass all test cases in https://github.com/WebAssembly/spec/blob/main/test/core/simd/simd_conversions.wast.

liutao-liu, 2023-12-05:
> Would u mind providing the reproducing steps and release version information?
> 
> Btw, I run a quick test on above highlight case and got the right answer.
> 
> ```
> webassembly> i8x16.narrow_i16x8_u 0x0\0x0 0xffffffffffffffff\0xffffffffffffffff
> <0x0000000000000000 0x0000000000000000>:v128
> ```

test.wat
```
(module
  (func (export "i8x16.narrow_i16x8_u") (param v128 v128) (result v128)(i8x16.narrow_i16x8_u (local.get 0) (local.get 1)))
)
```

After compiling test.wat using wamrc, i got test.aot. But the result of running test.aot through iwasm is as follows:
```
 iwasm -f i8x16.narrow_i16x8_u  test.aot 0x0/0x0  0xffffffffffffffff/0xffffffffffffffff
<0x0000000000000000 0xffffffffffffffff>:v128
```

lum1n0us, 2023-12-08:
we all tried your cases with both iwasm+wamrc 1.2.3 and latest iwasm+wamrc, and we all can't reproduce the problem.

Our results like:

```
iwasm-1.2.3 -f i8x16.narrow_i16x8_u test.aot 0/0 -1/-1
<0x0000000000000000 0x0000000000000000>:v128
```

liutao-liu, 2023-12-16:
> we all tried your cases with both iwasm+wamrc 1.2.3 and latest iwasm+wamrc, and we all can't reproduce the problem.
> 
> Our results like:
> 
> ```
> iwasm-1.2.3 -f i8x16.narrow_i16x8_u test.aot 0/0 -1/-1
> <0x0000000000000000 0x0000000000000000>:v128
> ```

The runtime environment I am using is arm64+Ubuntu 20.4 LTS. After compiling with AOT LLVM, the problem can be reproduced 100%. I don't know why you can't reproduce the problem.

[image]

From the analysis of the simd proposal, these three simd arrow ops should be handled in a signed manner when dealing with overflow scenarios.According to the description in this proposal, my [PR](https://github.com/bytecodealliance/wasm-micro-runtime/pull/2850/files#diff-cd849d38c95f6cb7fb658700206cd7aa2871aef13d6ce7401817d7c2b5d77731) has been revised. Please review it:

lum1n0us, 2023-12-21:
*ARM64* is the reason why we can't reproduce locally. Thanks for reporting and fixing.

liutao-liu, 2024-01-05:
Got

closed 2024-01-05 by liutao-liu