wasm-micro-runtime/wasm-micro-runtime #2846: zephyr: unable to build simple example targeting the nucleo_f767zi board with wasi libc

deadprogram, 2023-12-02:
I am trying to build the `product-mini/platforms/zephyr/simple` example targeting the `nucleo_f767zi` board with `WAMR_BUILD_LIBC_WASI` so that a WASM module compiled for `wasi` can execute on the board itself.

This is the command I am trying to use to build:

```
ZEPHYR_BASE=~/zephyrproject/zephyr west build -b nucleo_f767zi . -p always -- -DWAMR_BUILD_TARGET=THUMBV7 -DWAMR_BUILD_LIBC_WASI=1
```
However, the code is unable to build without errors:

```
-- Zephyr version: 3.5.99 (/home/ron/zephyrproject/zephyr), build: zephyr-v3.5.0-2399-gc94cd30cecbf                                                                          [31/204] Building C object CMakeFiles/app.dir/home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c.obj            
FAILED: CMakeFiles/app.dir/home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c.obj                               
ccache /home/ron/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBH_FREE=wasm_runtime_free -DBH_MALLOC=wasm_runtime_malloc -DBH_PLATFORM_ZEPHYR -DBUILD_TARGET=\"
THUMBV7\" -DBUILD_TARGET_THUMB -DCORE_CM7 -DHSE_VALUE=8000000 -DKERNEL -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32F767xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -DWASM_DISABLE_H
W_BOUND_CHECK=0 -DWASM_DISABLE_STACK_HW_BOUND_CHECK=0 -DWASM_DISABLE_WAKEUP_BLOCKING_OP=0 -DWASM_ENABLE_AOT=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWASM_ENABLE_FAST_INTERP=0 -DWASM_
ENABLE_GLOBAL_HEAP_POOL=1 -DWASM_ENABLE_INTERP=1 -DWASM_ENABLE_LIBC_BUILTIN=1 -DWASM_ENABLE_LIBC_WASI=1 -DWASM_ENABLE_MINI_LOADER=0 -DWASM_ENABLE_MODULE_INST_CONTEXT=1 -DWAS
M_ENABLE_MULTI_MODULE=0 -DWASM_ENABLE_SHARED_MEMORY=0 -DWASM_GLOBAL_HEAP_SIZE=131072 -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_ST
ART -D__ZEPHYR__=1 -I/home/ron/Development/wasm-micro-runtime/core/iwasm/interpreter -I/home/ron/Development/wasm-micro-runtime/core/iwasm/aot -I/home/ron/Development/wasm-m
icro-runtime/core/iwasm/libraries/libc-builtin -I/home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/include -I/home/ron/Deve
lopment/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src -I/home/ron/Development/wasm-micro-runtime/product-mini/platforms/zephyr/simple/../
../../../core/iwasm/include -I/home/ron/Development/wasm-micro-runtime/core/shared/platform/zephyr -I/home/ron/Development/wasm-micro-runtime/core/shared/platform/zephyr/../include -I/home/ron/Development/wasm-micro-runtime/core/shared/platform/common/libc-util -I/home/ron/Development/wasm-micro-runtime/core/shared/mem-alloc -I/home/ron/Develop
ment/wasm-micro-runtime/core/iwasm/common -I/home/ron/Development/wasm-micro-runtime/core/shared/utils -I/home/ron/zephyrproject/zephyr/include -I/home/ron/Development/wasm-
micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/include/generated -I/home/ron/zephyrproject/zephyr/soc/arm/st_stm32/stm32f7 -I/home/ron/zephyrproject/zephyr/
include/zephyr/posix -I/home/ron/zephyrproject/zephyr/lib/posix/getopt/. -I/home/ron/zephyrproject/zephyr/drivers -I/home/ron/zephyrproject/zephyr/soc/arm/st_stm32/common -I
/home/ron/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/ron/zephyrproject/zephyr/modules/cmsis/. -I/home/ron/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx
/soc -I/home/ron/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include -I/home/ron/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include/Legacy -
I/home/ron/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -isystem /home/ron/zephyrproject/zephyr/lib/libc/common/include -std=gnu99 -ffunction-sections -fdata-
sections                                          -Wall -Wno-unused-parameter -Wno-pedantic -mthumb -fno-strict-aliasing -Os -imacros /home/ron/Development/wasm-micro-runtim
e/product-mini/platforms/zephyr/simple/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7
 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/ron/zephyr-sdk-0.16.4/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/ron/zephyrproject/zephyr/include/zephy
r/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -W
error=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/
home/ron/Development/wasm-micro-runtime/product-mini/platforms/zephyr/simple=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ron/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-ma
p=/home/ron/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -D_POSIX_THREADS -std=c99 -MD -MT CMakeFiles/app.dir/home/ron/Development/wa
sm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c.obj -MF CMakeFiles/app.dir/home/ron/Development/wasm-micro-runtime/core/iwasm/librar
ies/libc-wasi/sandboxed-system-primitives/src/random.c.obj.d -o CMakeFiles/app.dir/home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-pr
imitives/src/random.c.obj -c /home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c
/home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c: In function 'open_urandom':
/home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c:77:15: warning: implicit declaration of function 'open'; did
 you mean 'fopen'? [-Wimplicit-function-declaration]
   77 |     urandom = open("/dev/urandom", O_RDONLY);
      |               ^~~~
      |               fopen
/home/ron/Development/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c:77:36: error: 'O_RDONLY' undeclared (first use in this funct
ion)
   77 |     urandom = open("/dev/urandom", O_RDONLY);
      |                                    ^~~~~~~~
....
```

Any ideas? Thanks in advance!

deadprogram, 2023-12-02:
If I try to execute my WASM code compiled targeting wasi I get the following log from the board:

```
Connected to /dev/ttyACM0. Press Ctrl-C to exit.
*** Booting Zephyr OS build zephyr-v3.5.0-2399-gc94cd30cecbf ***
WASM module instantiate failed: allocate memory failed
elapsed: 14
```

deadprogram, 2023-12-02:
If I compile `iwasm` for Linux, and I set `WAMR_BUILD_LIBC_WASI 0` then I get the following error:

```
$ ./iwasm ../../../app-samples/hello-world-tinygo/test.wasm 
[16:13:34:719 - 7FCAE7617B80]: warning: failed to link import function (wasi_snapshot_preview1, fd_write)
Exception: lookup the entry point symbol (like main, _main, __main_argc_argv) failed
```

If I compile `iwasm` with `WAMR_BUILD_LIBC_WASI 1` then the code executes as I expect:

```
$ ./iwasm ../../../app-samples/hello-world-tinygo/test.wasm 
Hello world!
```

hasheddan, 2023-12-04:
@deadprogram thanks for reporting! I've been working on the zephyr support and will look into this!

deadprogram, 2024-01-09:
Hello @hasheddan any updates on this by any chance? :smile_cat:

deadprogram, 2024-01-25:
:wave: :smiley_cat:

hasheddan, 2024-01-25:
@deadprogram thank you for the ping and my apologies on the delay here! I have just opened #3086 to address the primary issue you encountered here, but am also investigating additional rough spots regarding WASI support with Zephyr. Feel free to try out the patch and report any further problems you encounter!

salva00, 2024-01-25:
Hello, I have the same issue with build to qemu_cortex_a53 :(
I am running the latest wamr version (I've just pulled from main branch)

srberard, 2024-03-21:
Is there any update on this issue?  I'm having similar problems building with `WAMR_BUILD_LIBC_WASI 1` on Zephyr.  I'm still targeting WASI preview 1 so I'm on the `release/1.3.x` branch.