wasm-micro-runtime/wasm-micro-runtime #2847: Different output number when running a same binary

XinyuShe, 2023-12-02:
I used the AOT mode of different runtimes to run randomly generated wasm binaries, and the output number from WAMR was different from the others.
[filea1461_4.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13533925/filea1461_4.zip)
[image]

```bash
root@4252f5ec38df:/a# wamrc --bounds-checks=1 -o filea1461_4.wasm.aot filea1461_4.wasm ; iwasm --heap-size=0 -f main filea1461_4.wasm.aot
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file filea1461_4.wasm.aot was generated.
0x1:i32,0x1:i32,0x1:i32,0x1:i32,0x1:i32,0x1:i32

root@4252f5ec38df:/a# iwasm --heap-size=0 -f main filea1461_4.wasm    
0x1:i32,0x1:i32,0x1:i32,0x1:i32,0x1:i32,0x1:i32

root@4252f5ec38df:/a# wasmtime_16 compile filea1461_4.wasm -o filea1461_4.wasm.aot ; wasmtime_16 --allow-precompiled --invoke main filea1461_4.wasm.aot
warning: using `--invoke` with a function that returns values is experimental and may break in the future
65536
65536
65536
65536
65536
65536

root@4252f5ec38df:/a# wasmtime_16 --invoke main filea1461_4.wasm    
warning: using `--invoke` with a function that returns values is experimental and may break in the future
65536
65536
65536
65536
65536
65536
```

TianlongLiang, 2023-12-05:
Hi, thanks for finding this corner case and letting us know. I created a minimum reproducible case for your case:
```wast
(module
  (type (;0;) (func (result i32 i32 i32 i32 i32 i32)))
  (func (;0;) (type 0) (result i32 i32 i32 i32 i32 i32)
    (local i32 f64)

    i32.const 0
    memory.grow
    local.set 0

    local.get 0
    local.get 0
    local.get 0
    local.get 0
    local.get 0
    local.get 0
    return
    )
  (memory (;0;) 65536 65536)
  (export "main" (func 0))
  (data (;0;) (i32.const 7814725) "MN,_u\d2\a8\09P$\1c\0bHl.\04C\0e\05mF?\00\1b!GT\d1\9b\d2\ac=\185\d9\bawI\22gx3v8\07qt\c5\bf\05\0bz\16{h>\5cY\01A2\de\839JU5%L\01\08] \0da9P\1eZ5Z-BQW\01#uhr\0a\c8\9e1\02JoQ\07\13cv ^\10\0f gCR2\22/@\d6\94i\22\5c\13\1a #'*v\15*\5c\16M,\c9\88'I?a\0c(S\18_Zn\0f\7f;G\0amcQ\d1\bfeU\7frI/mJp\1e\1ab\18FXRC\12QK\18s/puyX\1e2U\de\a1t{ >\d4\b6\1b\01\12\11\193\14E\09[b\03i\12_\ce\a3AJK\10\0fR|.<C\17k&4`wS\16II\03gz#\10\17J\00T\1c\1f,z\05\17r\0d>\19qwx|~]\13Ac5z%vY\d5\a3:y\08\7f\07\1c\0b\10aKFEUmE,\01C\02Bs\06\02E\5cW\de\b5-FFmpd8M%\02dcv`2\06BXvq\15\15K(%\13G*0\1bzY/|YM`\18\01C<UtW\1bt#=\16K:)zf\05J\e7\ab\b7;Jzs%GO\14\df\9c\1ck=2rs\02v\1eC\1f*\16m\0bEB0\18;H\d6\b8}aiK\12Q}\12\d2\b2/W\e2\8a\80=Q\cf\8eBzxj\0cG}'@(n\c6\9a\22\0e92\1fV#:NE@=l2_!U3\13;(C8@A\c8\9eI\7fi*\01G5Ri\e9\b0\a5f\15\15\0c5\0c9O\0a;m\139wvk@\09~y2\11wyIC`W\0c\08\17E~t0L\09]j0\5cRv\1d\0b*\02mr\07My$@\17\d1\b4\1c\15a^\07\c9\bc,>\1aG\13j,OD\d7\966_\0b\22.},1RT8{9\06\c5\9d[3~+V^~v\c2\bd\7fk\0b\088YaN~n4J2@8/}[J(\006\11%}\1b^jsit\18\1b>/i\03:\1d1Ol\08k\14Rw8H 9AeG\04\0e\10)mOF:\06&\11Kd\10\d5\8bB\0a6~Q\04e?c \e8\a5\bb\05K\0aY\0a\19\22\1fK-\e9\a5\92>2 ig^$\04a\00_L3$\d7\88'\c4\b86i=\cf\83qV}\13P\11oK_\019m?S \1fVdMwX\10u\ed\9e\9fp\1aYQfzlNm=i\043\119[U067\063 \11l\11Il\13K^7KHh1BMu+^\19Ba6I\09>\14~V\0b`\5c\1c\1b\08\19Ef\5c \09]~H_\19\0e;s\06\04Gy(\0fP\11\22L\7f\0cS%a\0c\05dA.wI<Z\17a3|{ES\7f+Wi0\5c\03J.Dzwi\0d\0e\06\05\18]\02qbLpS\055ag\1f8K\16_\06}\06*B\09B1\14ggq\d6\9ex`\c9\bb\1d(U\e1\b0\b6yS%~@'\22t6+zXN\06\01\1f@<I\0c\07;\1fV\e3\a4\b1z\08 n\1b2\d4\8fi|\db\85\1ew+;EG\15\1d\03\07c`Q\0bf\1c\13\1di\7f2*\10X*-\cd\948H>z4\cf\b6\14&\c2\91/1J\10V\16\0d<-z\08a2\1c\0b\0fDZ\08tS\dc\b3&\1c\0c\14\14wL\df\b1\1b\044\1f!v\17T\c8\83FFf\e7\88\8cNv6\00*=\08B\19\07/P\07\19PGE\02\01\1fuB2\22\08#!\01OXB`ec{A}\c5\a2kj\05H)u\22\10t/UJlUS\0f\01\cf\96\06T \09A:r\02yOvSBpr\1exs\17xs5_y)'g_|k\12Lj\19\17K3'r\11D'\12\0d-=Gd\0fi#oZ.\00:b6JY\16/uSw\0dP2[\00\10.\0c\04`\06\5c(\16Q\e6\bd\b9l=[t[2&[a\16\5cI\10Y\0e\00\11\19B(|&Z5h^@RDA\7fw\03\06U\0d=\22o\5c\04yC\229\1e\02fo\1d?\c5\b5\cd\913\d9\90Ab%YR?HA\dd\aa9Dx/Jq>\18\1cFRLv\0c\18;\dc\87okqR4\ca\aee1\16\1aoq\1e\1b~h\d6\abU\dd\b0a\03\1e\0eMX\0f\1f0A\14X7\157E\17\03z{\07\d8\a7\ce\bcp1&R+\df\9c(P6RKy\08if\11Zt\00w/)q?5\cc\8a\00/\00Z+Ox:HNL_2vo\04jIp9\c3\be(\0dv~?b\04{ f|9\0eKl\13._\d0\be>plc\14L\021e\1c11A?\7f!\03G7\c9\bd\d0\9a7&2x\7f\03385\d5\94*x!\11;\06\06Z*\0a\10*(p8PaqHW2\05\ca\99\7fE3YO*@T\13<n\17r\0e+\18_\08\0f\1d-H#b\ec\90\a84TCL\c3\be\c7\95H\09\0fCd8\18rK\ca\b9A\d0\b0\01K\10W\07a\0b=p/UN$;:gR\03_z\08\11X6`I\0d_0.\0c\d6\862C\13&\17p^f\0a<k}55{2xAr\0dnEfoo\0c{cbO*N\0c_|!\11\0f\0c\1e\0d\5c]l\1f5#B&li+?\1e*H\13P8^\7f6\0d\03t&\05\17~\ce\9e]L\d0\adX\1dn\e9\88\a3)\05}3\05!G*\e9\8a\9c]#_B\cf\91\03\14Xl-F'9*a*]i>+5G40Wo%1\15$0tK_\11\0dh\13'RM\10\22:o\15%.\05e{sR\22z5=am@|$\00;\1aH(u\c4\94\147\15D;,)\1d\096\14o/\17>v\08\14\1d19D-\00E0\04*%\d2\a42F \0b\e6\88\aa@\0bU)\039S!s/3]_\0fbr,\18\1b\06S\12\0f5g{\1aT\01\d0\a8\dc\9f@LZJB\14,vm-8-{\0f$%R_|C8 <0h\c9\a2A)M]v\d8\aewL6SD7\d9\bc\0a\11tH\0dbCb\05eJ/hK+\1asE1\0b4]pl+#a:{AE\1f\0e\c2\a4(^}M\0a\04=\5c%}k33\1dm\1ea\db\938\1f)\1eU7,2//n[!\c5\ae\00\03\7f\080Kb\1c)O9\15u\17`-\17%ZDI\17\1e\1cwM\1e \11@N5gT\cb\ba:4\01^\09XbTF7\0biI\1c0\15B#\12?Z3Dh\e9\85\b0jbRc*e{\01\055^@sXBbg\0ag\0a(+\08\03*H\18\22]7j\02U\5c\12v\18H=\17KN\df\abCdq\c8\9d^So~\14O@|`Z\06|x\1f/B\1aC|J\1b#-O\1be\0bA>\0c?Nq\0a\03j\22\05\cb\acKCrz\13I7E\7f\14lwK!N\06k30<\10B%g;c\0bd\01!\05*&v\13\c5\91BG&FJ\1cJ\10D\07o\14Sp'I\11DP&\07v=J'r|{P\00HEg\11L1tra*~\07d4\055jR%^\07:\13\0bf)\07\19uls}6sPv\1aI5\11z\088i\1dhR\dc\a6\1f\03{E\1aypD\15@vX\1bU?\02x\10W\18H$j.\22\dd\94+\12E\10Nf\eb\a3\8dZ3\ce\a8z0\d5\ac)l\cd\b8}f\c7\a2@]^\0a$\10\16\1fwDr\14yG\00d\02\10!\1f&E\16}_\11kZ\d9\9e\17y>h\06T2;\03;\0a\15E*Zv\ce\a6k\09|T6\15Z\1a\19q:8\e3\ab\94\226\1c\1c\e2\95\a8_\08\1c\de\8fmAo?N\16O9H\e7\b3\8e=m7o\1bP\0dm\0eLM|\0bGB\1f_p.F\1b\112h9V{\c4\86I\10\5c}\05\1aQ30$i_\c6\b3\0b\03\0f1Ub\09= X\1e\00\cc\8d\e4\86\93%s\0a8\ca\bf\14`\00O:\1bb\1e-[\05e\cc\9f;3\1c\05\13Zqe8\09\cd\99\c2\b7}v\1e@H\0dX\14\c7\92\c8\81\17\14\c6\a1\c4\a4RC(`w)t=&h-d[\0c;!<X,nNpS*3SZ\c9\93F_F\1a\cf\abl67\c8\a5m\d3\bd\dc\b8m=2{\08Um\12\7f\1fVk\05.a\1a\5cN\18\1f3l\0c\032\0a\12!\0c#\1a\1c\13Q\11:{;V\18T'\04QAp\0cCBkH\15.|\15L*v=7\11T&*nb\06c\14XOa)u\d2\a9\5cA\07\de\a7\05G\0dS3%O\023i\05"))

```

The difference in return values comes from a current implementation limitation of WAMR, which is how WAMR handles the situation where the initial memory page count reaches the maximum. For the consideration of performance, WAMR allocates a "huge page" instead of the maximum number of pages(65536). This allocation method explains the occurrence of 0x1. We are working on it and hope we can find an elegant way to this limitation soon. We will keep you updated on our progress.

wenyongh, 2024-03-15:
The issue should have been resolved with #3209, see also #2468, #2530.