wasm-micro-runtime/wasm-micro-runtime #2858: Different output when running a same binary

abc767234318, 2023-12-04:
I used the interpreter mode of different runtimes to run randomly generated wasm binaries, and the output from wamr is different from the other runtimes.
[3605.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13543771/3605.zip)

```
iwasm --heap-size=0 -f 4 3605.wasm 1
```

```
-----------------wamr-----------------
WASM module load failed: type mismatch: br_table targets must all use same result type

-----------------wasmedge-----------------
[2023-11-22 22:19:18.333] [error] execution failed: out of bounds memory access, Code: 0x88
[2023-11-22 22:19:18.333] [error]     Accessing offset from: 0xb1a6c5f6 to: 0xb1a6c5f7 , Out of boundary: 0x000fffff
[2023-11-22 22:19:18.333] [error]     In instruction: i64.load16_s (0x32) , Bytecode offset: 0x00000f7e
[2023-11-22 22:19:18.333] [error]     When executing function name: "4"

```

wenyongh, 2023-12-05:
@abc767234318 Thanks for spotting the issue! Please try #2864.

wenyongh, 2024-01-05:
Close this issue as it was fixed.

closed 2024-01-05 by wenyongh