wasm-micro-runtime/wasm-micro-runtime #2865: Different output number when running a same binary

XinyuShe, 2023-12-05:
I used the AOT mode of different runtimes to run randomly generated wasm binaries, and the output number from WAMR was different from the others
[filea4199_8.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13556867/filea4199_8.zip)

```bash
==================================================
/a/filea4199_8.wasm
-----------------wasmtime-----------------
Error: failed to run main module `/a/filea4199_8.cwasm`

Caused by:
    0: failed to invoke `main`
    1: error while executing at wasm backtrace:
           0:  0x39f - <unknown>!<wasm function 0>
           1: 0x37c4 - <unknown>!<wasm function 33>
    2: wasm trap: integer divide by zero

-----------------wasmer-----------------
Compiler: cranelift
Target: x86_64-unknown-linux-gnu
✔ File compiled successfully to `/a/filea4199_8.wasmu`.
error: failed to run `/a/filea4199_8.wasmu`
╰─▶ 1: RuntimeError: integer divide by zero
           at <unnamed> (/a/filea4199_8.wasm[0]:0x39f)
           at <unnamed> (/a/filea4199_8.wasm[33]:0x37c4)

-----------------wamr-----------------
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file /a/filea4199_8.wasm.aot was generated.
Exception: out of bounds memory access

-----------------wasmedge-----------------
[2023-11-30 14:10:55.470] [info] compile start
[2023-11-30 14:10:55.477] [info] verify start
[2023-11-30 14:10:55.480] [info] optimize start
[2023-11-30 14:10:55.582] [info] codegen start
[2023-11-30 14:10:55.661] [info] output start
[2023-11-30 14:10:55.684] [info] compile done
[2023-11-30 14:10:55.789] [error] execution failed: integer divide by zero, Code: 0x84
[2023-11-30 14:10:55.789] [error]     When executing function name: "main"
```
```bash
/home/wasm-micro-runtime/wamr-compiler/build/wamrc --version:
	wamrc 1.2.3

----------------------------------------------------
/home/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --version:
	iwasm 1.2.3
```

```bash
wamr_command = f"{wamrc} --bounds-checks=1 -o {filename}.aot {filename}\
            ; {iwasm} --heap-size=0 -f main {filename}.aot"
```

TianlongLiang, 2023-12-05:
Hi, this should be due to the same reason as https://github.com/bytecodealliance/wasm-micro-runtime/issues/2847. I will test it again once the modifications for https://github.com/bytecodealliance/wasm-micro-runtime/issues/2847 are made.

TianlongLiang, 2024-02-02:
Update: this is because currently, WAMR has an optimization that resizes 65536 pages to 1 page that has 4GB size, so memory.size will return 1 instead of 65536, which is the reason for the discrepancy in this test case.