wasm-micro-runtime/wasm-micro-runtime #2889: Read directory bug ?

Userzxcvbvnm, 2023-12-08:
## Steps to reproduce
There is a file `Data/mydir/test.txt` is in the path `Data/mydir`.
The test program is to print the files in the path `Data/mydir`.

The c file is uploaded as 
[readdir-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/13613310/readdir-c.txt)

The wasm file is uploaded as 
[readdir-wasm.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/13613312/readdir-wasm.txt)

## Expected behavior
Print:
```
Successfully get the file fd.
.
..
test.txt
```

wasmer wasmtime wazero print the above message.

## Actual behavior
WAMR prints:
```
Successfully get the file fd.
opendir.
```

WAMR print the content of `Data/mydir` fails.

# Envrionment
Windows 10 (amd64)
WAMR-1.2.3

yamt, 2023-12-12:
on macOS, it worked as expected. maybe something specific to windows?

wenyongh, 2023-12-12:
Windows uses libc-uvwasi by default, the result may be different.