wasm-micro-runtime/wasm-micro-runtime #2901: what does `touch_pages` do and why it received a SIGSEGV?

ssssssyy0112, 2023-12-12:
I am using wamr in a brpc server, and when I call the server, the thread received a SIGSEGV like this:
[image]
It seems like during execution of wasm_runtime_full_init(), some thing wrong occured.
My code inside the brpc server impl is just like this, using Alloc_With_Allocator
```cpp
char *buffer, error_buf[128];

    wasm_module_t module = NULL;
    wasm_module_inst_t module_inst = NULL;
    wasm_exec_env_t exec_env = NULL;
    uint32 buf_size, stack_size = 8092, heap_size = 8092;
    wasm_function_inst_t func = NULL;

    RuntimeInitArgs init_args;
    memset(&init_args, 0, sizeof(RuntimeInitArgs));
    uint32_t arg[10];
    char* wasm_path = "../wasm/wasmcode.wasm";
    init_args.mem_alloc_type = Alloc_With_Allocator;
    init_args.mem_alloc_option.allocator.malloc_func = (void*)malloc;
    init_args.mem_alloc_option.allocator.realloc_func = (void*)realloc;
    init_args.mem_alloc_option.allocator.free_func = (void*)free;

    if (!wasm_runtime_full_init(&init_args)) {
        printf("Init runtime environment failed.\n");
        return -1;
    }
```

ssssssyy0112, 2023-12-12:
I try  to insert some printf to help debug, and find something may be wrong during execution inside wasm_runtime_full_init near wasm_runtime_env_init(), but I couldn't figure out what happened.
[image]

wenyongh, 2023-12-12:
@ssssssyy0112 touch_pages is needed for the native stack boundary check with hardware trap mechanism, if it doesn't work for you, could you try disabling it with `cmake -DWAMR_DISABLE_HW_BOUND_CHECK=1`:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#disable-native-stack-boundary-check-with-hardware-trap

And when generating AOT file, please add `--stack-bounds-checks=1` for wamrc to enable the software check instead.

wenyongh, 2023-12-15:
Hi, sorry for the typo, please had better use `cmake -DWAMR_DISABLE_STACK_HW_BOUND_CHECK=1` to build iwasm, and use `wamrc --stack-bounds-checks=1` to compile the wasm file. This only disables the native stack boundary hardware check, but keeps the wasm memory access boundary hardware check, which may improve the performance. If using `cmake -DWAMR_DISABLE_HW_BOUND_CHECK=1` and `wamrc --bounds-checks=1`, both of them are disabled.

It is good to building wasm app with `--export=__heap_base and --export=__data_end`, but I am not sure why `Exception: native stack overflow` disappears after using them.

ssssssyy0112, 2023-12-18:
> Hi, sorry for the typo, please had better use `cmake -DWAMR_DISABLE_STACK_HW_BOUND_CHECK=1` to build iwasm, and use `wamrc --stack-bounds-checks=1` to compile the wasm file. This only disables the native stack boundary hardware check, but keeps the wasm memory access boundary hardware check, which may improve the performance. If using `cmake -DWAMR_DISABLE_HW_BOUND_CHECK=1` and `wamrc --bounds-checks=1`, both of them are disabled.
> 
> It is good to build wasm app with `--export=__heap_base and --export=__data_end`, but I am not sure why `Exception: native stack overflow` disappears after using them.

hmmmm， this time I found my program happend to throw exception `Exception: native stack overflow` , so could you plz tell me in what situations will this exception appear?

gnh1201, 2023-12-19:
The `DWAMR_DISABLE_HW_BOUND_CHECK=1` option does resolve the issue, but it seems reasonable to expect error messages that sufficiently explain the problem without applying this option. As you mentioned, relying solely on SIGSEGV is not enough to adequately explain the cause of the error.

In my case, I couldn't obtain useful information for problem resolution even with debuggers like GDB, so I resorted to using `printf` to pinpoint the problematic section.

There might be relevance to my previous work that could be associated with this issue. Please consider it as it might be helpful for reference.

* #1166
* #1058
* #802