wasm-micro-runtime/wasm-micro-runtime #2918: Exception: failed to call unlinked import function (libdyntype, dyntype_get_context)  on Android platform.

M201370367, 2023-12-22:
I met a exception when execute aot file on Android platform. Who can help me? Thanks a lot.
Exception:
`
12-22 17:33:09.956  9534  9534 I wasm_jni::: wasm_runtime_full_init
12-22 17:33:09.956  9534  9534 I wasm_jni::: use an internal test file, gona to output Hello World in logcat
12-22 17:33:09.956  9534  9534 I wasm_jni::: wasm_runtime_load file buf_size= 7876   4
12-22 17:33:09.956  9534  9534 I wasm_jni::: wasm_runtime_instantiate
12-22 17:33:09.957  9534  9534 I wasm_runtime::: [13:32:42:470 - EB61C470]: 
12-22 17:33:09.957  9534  9534 I wasm_runtime::: warning: failed to link import function (libdyntype, dyntype_get_context)
12-22 17:33:09.957  9534  9534 I wasm_runtime::: 
12-22 17:33:09.957  9534  9534 I wasm_runtime::: [13:32:42:544 - EB61C470]: 
12-22 17:33:09.957  9534  9534 I wasm_runtime::: warning: failed to link import function (libdyntype, dyntype_new_number)
12-22 17:33:09.957  9534  9534 I wasm_runtime::: 
12-22 17:33:09.957  9534  9534 I wasm_runtime::: [13:32:42:606 - EB61C470]: 
12-22 17:33:09.957  9534  9534 I wasm_runtime::: warning: failed to link import function (libdyntype, dyntype_new_string)
12-22 17:33:09.957  9534  9534 I wasm_runtime::: 
12-22 17:33:09.957  9534  9534 I wasm_runtime::: [13:32:42:667 - EB61C470]: 
12-22 17:33:09.957  9534  9534 I wasm_runtime::: warning: failed to link import function (env, Console_log)
12-22 17:33:09.957  9534  9534 I wasm_runtime::: 
12-22 17:33:09.957  9534  9534 I wasm_jni::: run main() of the application
12-22 17:33:09.957  9534  9534 I wasm_jni::: wasm_runtime app_instance_main
12-22 17:33:09.957  9534  9534 I wasm_jni::: Exception: failed to call unlinked import function (libdyntype, dyntype_get_context)
12-22 17:33:09.957  9534  9534 I wasm_jni::: wasm_runtime_deinstantiate
12-22 17:33:09.957  9534  9534 I wasm_jni::: wasm_runtime_unload
12-22 17:33:09.957  9534  9534 I wasm_jni::: wasm_runtime_destroy
`

android cmake I changed:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/android/CMakeLists.txt
`
# Copyright (C) 2019 Intel Corporation.  All rights reserved.
# SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception

cmake_minimum_required (VERSION 3.4.1)

set (CMAKE_VERBOSE_MAKEFILE on)
set (CMAKE_BUILD_TYPE       Release)

# https://github.com/actions/virtual-environments/blob/main/images/linux/Ubuntu2004-Readme.md#environment-variables-3
set (CMAKE_TOOLCHAIN_FILE "/home/mi/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake")
set (ANDROID_SDK "/home/mi/Android/Sdk")
set (ANDROID_NDK "/home/mi/Android/Sdk/ndk/23.1.7779620")

set (ANDROID_ABI "armeabi-v7a")
set (ANDROID_LD lld)
if (NOT DEFINED ANDROID_PLATFORM)
  set (ANDROID_PLATFORM 24)
endif ()

project (iwasm)

set (WAMR_BUILD_PLATFORM      "android")
set (WAMR_BUILD_TARGET        "ARMV7")
set (WAMR_BUILD_TYPE          Release)
set (WAMR_BUILD_INTERP        1)
set (WAMR_BUILD_AOT           1)
set (WAMR_BUILD_JIT           0)
set (WAMR_BUILD_LIBC_BUILTIN  1)
set (WAMR_BUILD_LIBC_WASI     1)
set (WAMR_BUILD_GC_BINARYEN   1)
set (WAMR_BUILD_FAST_INTERP   1)
set (WAMR_BUILD_GC            1)
#set (WAMR_BUILD_STRINGREF 1)
add_definitions(-DWASM_TABLE_MAX_SIZE=10240)

if (NOT DEFINED WAMR_BUILD_SIMD)
  # Enable SIMD by default
  set (WAMR_BUILD_SIMD 1)
endif ()

# Reset default linker flags
set (CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
set (CMAKE_SHARED_LIBRARY_LINK_CXX_FLAGS "")

# Set WAMR_BUILD_TARGET, currently values supported:
# "X86_64", "AMD_64", "X86_32", "AARCH64[sub]", "ARM[sub]", "THUMB[sub]", "MIPS", "XTENSA"
if (NOT DEFINED WAMR_BUILD_TARGET)
  if (CMAKE_SIZEOF_VOID_P EQUAL 8)
    # Build as X86_64 by default in 64-bit platform
    set (WAMR_BUILD_TARGET "X86_64")
  elseif (CMAKE_SIZEOF_VOID_P EQUAL 4)
    # Build as X86_32 by default in 32-bit platform
    set (WAMR_BUILD_TARGET "X86_32")
  else ()
    message(SEND_ERROR "Unsupported build target platform!")
  endif ()
endif ()

if (NOT CMAKE_BUILD_TYPE)
  set(CMAKE_BUILD_TYPE Release)
endif ()

if (NOT DEFINED WAMR_BUILD_INTERP)
  # Enable Interpreter by default
  set (WAMR_BUILD_INTERP 1)
endif ()

if (NOT DEFINED WAMR_BUILD_AOT)
  # Enable AOT by default.
  set (WAMR_BUILD_AOT 0)
endif ()

if (NOT DEFINED WAMR_BUILD_JIT)
  # Disable JIT by default.
  set (WAMR_BUILD_JIT 0)
endif ()

if (NOT DEFINED WAMR_BUILD_LIBC_BUILTIN)
  # Enable libc builtin support by default
  set (WAMR_BUILD_LIBC_BUILTIN 1)
endif ()

if (NOT DEFINED WAMR_BUILD_LIBC_WASI)
  # Enable libc wasi support by default
  set (WAMR_BUILD_LIBC_WASI 1)
endif ()

set (WAMR_ROOT_DIR ${CMAKE_CURRENT_LIST_DIR}/../../../../../deps/wamr-gc)
#include_directories(${WAMR_ROOT_DIR}/core/iwasm/include)
#include_directories(${WAMR_ROOT_DIR}/core/iwasm/common/gc)

include (${WAMR_ROOT_DIR}/build-scripts/runtime_lib.cmake)

set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--gc-sections -pie -fPIE")

set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -Wextra -Wformat -Wformat-security")
# set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wconversion -Wsign-conversion")

if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
  if (NOT (CMAKE_C_COMPILER MATCHES ".*clang.*" OR CMAKE_C_COMPILER_ID MATCHES ".*Clang"))
    set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
  endif ()
endif ()

# The following flags are to enhance security, but it may impact performance,
# we disable them by default.
#if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
#  set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ftrapv -D_FORTIFY_SOURCE=2")
#endif ()
#set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fstack-protector-strong --param ssp-buffer-size=4")
#set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wl,-z,noexecstack,-z,relro,-z,now")

if (USE_SANITIZER EQUAL 1)
    message("Sanitizer enabled")
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fsanitize=signed-integer-overflow  \
                        -fsanitize=undefined -fsanitize=address  \
                        -fno-sanitize=bounds,bounds-strict,alignment \
                        -fno-sanitize-recover -Wall -Werror -Wformat")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fsanitize=signed-integer-overflow  \
                        -fsanitize=undefined -fsanitize=address  \
                        -fno-sanitize=bounds,bounds-strict,alignment \
                        -fno-sanitize-recover -Wall -Werror -Wformat")
endif ()

## WAMR
include(${CMAKE_CURRENT_LIST_DIR}/../../../../../wamr_config.cmake)

## quickjs
set(QUICKJS_SRC_DIR ${CMAKE_CURRENT_LIST_DIR}/../../../../../deps/quickjs)

include_directories(${QUICKJS_SRC_DIR})

set(QUICKJS_SOURCE
    ${QUICKJS_SRC_DIR}/cutils.c
    ${QUICKJS_SRC_DIR}/libregexp.c
    ${QUICKJS_SRC_DIR}/libunicode.c
    ${QUICKJS_SRC_DIR}/quickjs.c)

## libdyntype
set(LIBDYNTYPE_DIR ${CMAKE_CURRENT_LIST_DIR}/../../../../../libdyntype)
include (${LIBDYNTYPE_DIR}/libdyntype.cmake)

## stdlib
set(STDLIB_DIR ${CMAKE_CURRENT_LIST_DIR}/../../../../../stdlib)

include_directories(${STDLIB_DIR})

set(STDLIB_SOURCE
    ${STDLIB_DIR}/lib_console.c
    ${STDLIB_DIR}/lib_array.c
    ${STDLIB_DIR}/lib_timer.c
)

## struct-indirect
set(STRUCT_INDIRECT_DIR ${CMAKE_CURRENT_LIST_DIR}/../../../../../struct-indirect)

include_directories(${STRUCT_INDIRECT_DIR})

set(STRUCT_INDIRECT_SOURCE
    ${STRUCT_INDIRECT_DIR}/lib_struct_indirect.c
)

## utils
set(UTILS_DIR ${CMAKE_CURRENT_LIST_DIR}/../../../../../utils)

include_directories(${UTILS_DIR})

set(TYPE_UTILS_SOURCE
    ${UTILS_DIR}/type_utils.c
)

set(OBJECT_UTILS_SOURCE
    ${UTILS_DIR}/object_utils.c
)

set(WAMR_UTILS_SOURCE
    ${UTILS_DIR}/wamr_utils.c
)
# Ignore warnings of QuickJS
set_source_files_properties(
    ${QUICKJS_SOURCE}
    PROPERTIES
    COMPILE_FLAGS "-w"
)

set(STRINGREF_DIR ${CMAKE_CURRENT_LIST_DIR}/../../../../../stringref)
set(WAMR_STRINGREF_IMPL_SOURCE
        ${STRINGREF_DIR}/stringref_qjs.c
        )

include (${SHARED_DIR}/utils/uncommon/shared_uncommon.cmake)

find_library(
        log-lib
        log)
find_library(android-lib android)

add_library (iwasm SHARED
    ${WAMR_RUNTIME_LIB_SOURCE}
    ${UNCOMMON_SHARED_SOURCE}
    ${QUICKJS_SOURCE}
    ${LIBDYNTYPE_SRC}
    ${STDLIB_SOURCE}
    ${STRUCT_INDIRECT_SOURCE}
    ${TYPE_UTILS_SOURCE}
    ${OBJECT_UTILS_SOURCE}
    ${WAMR_STRINGREF_IMPL_SOURCE}
    ${WAMR_UTILS_SOURCE}

    "${CMAKE_CURRENT_LIST_DIR}/wasm-jni.cpp")

    target_link_libraries(
            iwasm
            ${log-lib}
            ${android-lib})

if (CMAKE_BUILD_TYPE STREQUAL Release)
target_link_libraries (iwasm ${LLVM_AVAILABLE_LIBS} ${UV_A_LIBS} -lm -ldl -landroid -llog -s)
else()
target_link_libraries (iwasm ${LLVM_AVAILABLE_LIBS} ${UV_A_LIBS} -lm -ldl -landroid -llog)
endif()

set (distribution_DIR ${CMAKE_BINARY_DIR}/distribution)
set_target_properties (iwasm PROPERTIES LIBRARY_OUTPUT_DIRECTORY "${distribution_DIR}/wasm/lib")
`

M201370367, 2023-12-23:
I know the reason.

closed 2023-12-23 by M201370367

annaamodio, 2024-01-27:
similar problem, no idea. What was the reason? (thanks :))

xujuntwt95329, 2024-01-27:
@annaamodio Have you registered these symbols when initializing the runtime? You can refer to these lines:
https://github.com/web-devkits/Wasmnizer-ts/blob/355cd3d7eebbaa8f8033ec0938862592ed465333/runtime-library/main.c#L734-L771