wasm-micro-runtime/wasm-micro-runtime #2920: Unexpected exception in JIT mode and fast JIT mode

erxiaozhou, 2023-12-24:
# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake -DWASM_ENABLE_AOT=1 -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=1   -DCMAKE_INSTALL_PREFIX="xxx" -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1  -Bbuild
cd build
make -j40
```
or 
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWASM_ENABLE_AOT=1 -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=0   -DCMAKE_INSTALL_PREFIX="xxx" -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1 -Bbuild
```

execute:
```shell
iwasm --heap-size=0 --llvm-jit -f to_test <tc_name>
```
# Cases
[two_cases.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13762507/two_cases.zip)

# Actual behavior:
No exception

# Expected behavior:
An exception indicates that "WASM module load failed: undeclared function reference"

TianlongLiang, 2024-01-03:
Hi, could you please try it on the latest main branch? It now work as expected on my machine:
[image]

TianlongLiang, 2024-01-03:
Hi, this pr https://github.com/bytecodealliance/wasm-micro-runtime/pull/2972/files should fix it, now it won't  have the unexpected exception:
[image]

wenyongh, 2024-01-05:
Close this issue as it was fixed.

closed 2024-01-05 by wenyongh