wasm-micro-runtime/wasm-micro-runtime #2931: Different behavior between fast-jit and llvm-jit

unicornt, 2023-12-26:
Hi, I found a case that can cause different behavior between `fast-jit` mode and `llvm-jit` mode.

## Test Case
[case.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13769653/case.zip)

## Result
Run `iwasm --fast-jit case.wasm` and `echo $?` to see the return code.
[image]

## Extra info
I believe the different return code is caused by `call 0` function at the end of `function 5`. `function 0` is `proc_exit` function.

## Versions and Environment
wamr version : 1.2.3
Operating system: Ubuntu 20.04
Architecture: x86_64

TianlongLiang, 2024-01-03:
Hi, https://github.com/bytecodealliance/wasm-micro-runtime/pull/2969 should fix this issue, thanks for reporting the issue! 

[image]

wenyongh, 2024-01-05:
Close this issue as it was fixed.

closed 2024-01-05 by wenyongh