wasm-micro-runtime/wasm-micro-runtime #2935: release 1.3.1 ?

yamt, 2023-12-28:
it seems some users are annoyed much by the bug fixed by https://github.com/bytecodealliance/wasm-micro-runtime/pull/2882.
it might make sense to make a patch release for the fix.
how do you think?

wenyongh, 2023-12-28:
Yes, agree to release 1.3.1, but just wondering whether we should wait until some other issues are fixed? And we are also planning to merge `dev/gc_refactor` to main, which is a huge change including the AOT ABI (AOT file format and others), after that we can resolve some other AOT ABI issues, do you have any concern? My suggestion is to release 1.3.1 recently, then continue to fix some other issues and release 1.3.2, and the merge `dev/gc_refactor` to main, and then we can release 2.0.0 in the future.

yamt, 2023-12-28:
> My suggestion is to release 1.3.1 recently, then continue to fix some other issues and release 1.3.2, and the merge `dev/gc_refactor` to main, and then we can release 2.0.0 in the future.

it makes sense.

wenyongh, 2024-01-08:
@yamt is there any patch to merge to main branch from you? I will do some test and create release 1.3.1 if things go well.

yamt, 2024-01-09:
> @yamt is there any patch to merge to main branch from you? I will do some test and create release 1.3.1 if things go well.

i don't have anything urgent to block 1.3.1.

wenyongh, 2024-01-12:
@yamt the release 1.3.1 was created: https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/WAMR-1.3.1.

yamt, 2024-01-12:
> @yamt the release 1.3.1 was created: https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/WAMR-1.3.1.

thank you.

closed 2024-01-12 by yamt