wasm-micro-runtime/wasm-micro-runtime #2939: 使用emcc编译.c文件时会报错，但能正常生成.wasm文件并运行正常，请问如何消除错误提示。

kamylee, 2023-12-29:
我在一个hello.c文件中申请了：
int foo(int a, int b);
void foo2(char * msg, char * buffer, int buf_len);

EMSCRIPTEN_KEEPALIVE
int test2(char* strParam ,int iParam)
{
	printf("p1:%s , p2:%d .\n",strParam,iParam);

	int c = foo(3, 3);
	printf("test2 add:%d .\n",c);

    char * msg = "hello";
    char buffer[100];
    foo2(msg, buffer, sizeof(buffer));   // call into native foo2()
    printf("test2 buffer:%s .\n",buffer);

	return 1;
	// return "from wasm test1 api";
}

使用命令：emcc -O3 hello.c -o hello.wasm -s STANDALONE_WASM --no-entry
[image]
不过可以正常运行输出结果：
[image]

`error: undefined symbol: foo (referenced by top-level compiled C/C++ code)
warning: Link with `-sLLD_REPORT_UNDEFINED` to get more information on undefined symbols
warning: To disable errors for undefined symbols use `-sERROR_ON_UNDEFINED_SYMBOLS=0`
warning: _foo may need to be added to EXPORTED_FUNCTIONS if it arrives from a system library
error: undefined symbol: foo2 (referenced by top-level compiled C/C++ code)
warning: _foo2 may need to be added to EXPORTED_FUNCTIONS if it arrives from a system library
Error: Aborting compilation due to previous errors`
我要如何消除这个错误提示呢？

lum1n0us, 2023-12-30:
IIUC, `foo` and `foo2` **are** undefined symbols. what's your expectation on whole program?

kamylee, 2024-01-02:
我要在c++中加载这个wasm。foo和foo2的申明放在wasm文件中，foo和foo2定义在c++本地，我想在使用emcc编译时能以正确的方式编译，并不会报错。我不确定是不是我的操作有什么问题。因为虽然能编译出wasm并能正常运行，但感觉不完善，因为他是报过错的。
我要怎么写这个wasm文件，能正常调用外部函数，并在编译时不会报错？感谢！

lum1n0us, 2024-01-02:
in that case, [callback](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/wasm-c-api/src/callback.c) is an example to start with. Please be aware of that, there is no need to prepare [callback.wat](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/wasm-c-api/src/callback.wat). You should write a c++ program and use emcc to compile it to Wasm.

In C/C++, use [import_name](https://clang.llvm.org/docs/AttributeReference.html#import-name) and [import_module](https://clang.llvm.org/docs/AttributeReference.html#import-module) to declare a import function.