wasm-micro-runtime/wasm-micro-runtime #2941: [question] for embedded wasm application, is it support api like `select()/sleep or usleep()` ?

tkernelcn, 2023-12-29:
I didn't found select() api in socket sample code, is it support or how to support it?

Thanks.

lum1n0us, 2023-12-30:
last time I checked, `select`, `sleep` and `usleep` are all defined symbols in [wasi-libc](https://github.com/WebAssembly/wasi-libc/blob/main/expected/wasm32-wasi/defined-symbols.txt). It means all three have their webassembly implementations.

tkernelcn, 2023-12-30:
@lum1n0us thanks for your reply, I am new on wasm
currently, I already download wasi-sdk, and can build and run `hello world` on linux(wasm-micro-runtime/product-mini/platforms/linux/build)
but if I just add a `sleep(1);` in the demo, build will error:
```shell
$ /opt/wasi-sdk/bin/clang --target=wasm32 --sysroot=/opt/wasi-sdk/share/wasi-sysroot/ -O3  -nostdlib -z stack-size=32768  -Wl,--initial-memory=131072,--max-memory=131072 -Wl,--allow-undefined-file=/opt/wasi-sdk/share/wasi-sysroot/share/wasm32-wasi/defined-symbols.txt -Wl,--no-entry -Wl,--export-all -Wl,--export=__heap_base,--export=__data_end -Wl,--export=__wasm_call_ctors -D__wasi__ main.c -o test.wasm
main.c:15:5: error: call to undeclared function 'sleep'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
   15 |     sleep(5);
      |     ^
1 error generated.
```

would you please help to check what the problem is.
Thanks.

tkernelcn, 2023-12-30:
another question: can not export main, only can export all, very strange.
```shell
$ /opt/wasi-sdk/bin/clang --target=wasm32 --sysroot=/opt/wasi-sdk/share/wasi-sysroot/ -O3  -nostdlib -z stack-size=32768  -Wl,--initial-memory=131072,--max-memory=131072 -Wl,--allow-undefined-file=/opt/wasi-sdk/share/wasi-sysroot/share/wasm32-wasi/defined-symbols.txt -Wl,--no-entry -Wl,--export=main -Wl,--export=__heap_base,--export=__data_end -Wl,--export=__wasm_call_ctors -D__wasi__ main.c -o test.wasm
wasm-ld: error: symbol exported via --export not found: main
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
Thanks.

tkernelcn, 2023-12-30:
@lum1n0us  update
regarding the sdk api `sleep`, it works
```shell
$ /opt/wasi-sdk/bin/clang --target=wasm32-wasi --sysroot=/opt/wasi-sdk/share/wasi-sysroot/ -lrt -O3 -z stack-size=32768  -Wl,--initial-memory=131072,--max-memory=131072 -Wl,--allow-undefined-file=/opt/wasi-sdk/share/wasi-sysroot/share/wasm32-wasi/defined-symbols.txt -D__wasi__ main.c -o test.wasm
```
but also doesn't support export main

lum1n0us, 2023-12-30:
Here is a quick example:

```c
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

// __attribute__ is optional
__attribute__((export_name("main")))
int main() {
    printf("--> before sleep\n");
    sleep(3);
    printf("--> after sleep\n");
    printf("\n");
    return 0;
}
```

then

```bash
$ /opt/wasi-sdk-20.0/bin/clang-16 -o sleep_demo.wasm  main.c
```

> Usually, it isn't necessary to export `main()`. clang will give it a `_start()` as the entry point. If tends to export a function, use `__attribute__((export_name(NAME)))`

tkernelcn, 2023-12-31:
@lum1n0us crystal clear, thank you very much

closed 2023-12-31 by tkernelcn