wasm-micro-runtime/wasm-micro-runtime #2944: Unexpected exception on interpreter mode and JIT mode

erxiaozhou, 2023-12-31:
# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=0  -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1 -Bbuild
```
or
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_FAST_JIT=1 -DWAMR_BUILD_JIT=1 -DWAMR_ENABLE_LAZY_JIT=1  -DWAMR_BUILD_REF_TYPES=1 -DWAMR_ENABLE_BULK_MEMORY=1 -DWAMR_ENABLE_AOT=1 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1 -Bbuild
```
or
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DCMAKE_INSTALL_PREFIX="$install_dir" -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=0 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:
```shell
iwasm --heap-size=0 --llvm-jit -f to_test <tc_name>
```
or
```
iwasm --heap-size=0 --multi-tier-jit -f to_test <tc_name>
```
or
```
iwasm --heap-size=0 -f to_test <tc_name>
```
# Cases

[iwasm_no_undeclared_function_reference.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13800833/iwasm_no_undeclared_function_reference.zip)

# Actual behavior:
An exception
```
False WASM module load failed: undeclared function reference
```

# Expected behavior:
no exception

wenyongh, 2024-01-03:
Hi, thanks for spotting the issue. We fixed it with #2972.

wenyongh, 2024-01-05:
Close this issue as it was fixed.

closed 2024-01-05 by wenyongh