wasm-micro-runtime/wasm-micro-runtime #2945: unhandled SIGSEGV in fast interpreter mode

erxiaozhou, 2023-12-31:
# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:
```shell
iwasm --heap-size=0 -f to_test  <tc_name>
```
# Cases

[iwasm_fast_interp_moob_unhandled.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13800874/iwasm_fast_interp_moob_unhandled.zip)

# Actual behavior:

```
unhandled SIGSEGV, si_addr: (nil)
```

# Expected behavior:
An exception indicates that "out of bounds memory access"

xujuntwt95329, 2024-01-02:
Hi, thanks for reporting this.
We've fixed this issue in https://github.com/bytecodealliance/wasm-micro-runtime/pull/2971

erxiaozhou, 2024-01-02:
> Hi, thanks for reporting this. We've fixed this issue in #2971

Thank you for your repy!

wenyongh, 2024-01-05:
Close this issue as it was fixed.

closed 2024-01-05 by wenyongh