wasm-micro-runtime/wasm-micro-runtime #2948: Double free correption in fast interpreter mode

erxiaozhou, 2023-12-31:
# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:
```shell
iwasm --heap-size=0 -f to_test  <tc_name>
```
# Cases

[iwasm_fast_interp_int_overflow_double_free.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13800894/iwasm_fast_interp_int_overflow_double_free.zip)

# Actual behavior:

```
double free or corruption (out)
```

The issue is caused by the instruction i32.rem_s

# Expected behavior:
An exception indicates that "integer divide by zero"

wenyongh, 2024-01-02:
Hi, thanks for reporting the issue. It should have been fixed by #2971.

wenyongh, 2024-01-03:
Sorry, we found that uploaded file is same as file in #2947, it may be wrong? Would you mind uploading again?

erxiaozhou, 2024-01-03:
> Sorry, we found that uploaded file is same as file in #2947, it may be wrong? Would you mind uploading again?

I'm sorry that I uploaded the wrong file and I have updated it now. Thank you!

xujuntwt95329, 2024-01-04:
Thanks for reporting this, we've fixed in https://github.com/bytecodealliance/wasm-micro-runtime/pull/2974

TianlongLiang, 2024-01-05:
It seems that this file won't cause any exceptions? I use wasmtime to double-check this file, no exception is thrown

erxiaozhou, 2024-01-05:
> It seems that this file won't cause any exceptions? I use wasmtime to double-check this file, no exception is thrown

Sorry, I uploaded a wrong test case and I have updated it now.
The root cause seems to be the same to the issue https://github.com/bytecodealliance/wasm-micro-runtime/issues/2947
 thank you!

wenyongh, 2024-01-09:
Welcome and close this issue as it was fixed.

closed 2024-01-09 by wenyongh