wasm-micro-runtime/wasm-micro-runtime #2949: The runtime iwasm cause unexpected exception and unhandle SIGSEGV

erxiaozhou, 2023-12-31:
The test case triggers no exception on the classic interpreter mode, triggers an "unhandled SIGSEGV" on the fast interpreter mode, and triggers an exception indicating "integer divided by zero" on the fast jit mode.

# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

fast interpreter mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```
fast jit mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=0    -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1 -Bbuild
```

execute:
fast interpreter mode
```shell
iwasm --heap-size=0 -f to_test  <tc_name>
```
fast jit mode
```
iwasm --heap-size=0  --fast-jit -f to_test   <tc_name>
```
# Cases

[iwasm_fast_interp_int_overflow_double_free.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13800894/iwasm_fast_interp_int_overflow_double_free.zip)

# Actual behavior:

fast interpreter mode
```
unhandled SIGSEGV, si_addr: (nil)
Aborted
```
fast jit mode
```
Exception: integer divide by zero
```

# Expected behavior:
No exception

erxiaozhou, 2024-01-02:
The bug may be fixed by https://github.com/bytecodealliance/wasm-micro-runtime/pull/2971 and https://github.com/bytecodealliance/wasm-micro-runtime/pull/2969

wenyongh, 2024-01-05:
Close this issue as it was fixed.