wasm-micro-runtime/wasm-micro-runtime #2950: Unexpect excrption on fast  JIT mode

erxiaozhou, 2023-12-31:
# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

fast jit mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=0    -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1 -Bbuild
```

execute:

fast jit mode
```
iwasm --heap-size=0  --fast-jit -f to_test   <tc_name>
```
# Cases

[iwasm_fast_jit_unexpected_moob.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13801160/iwasm_fast_jit_unexpected_moob.zip)

# Actual behavior:

fast jit mode
```
Exception: integer divide by zero
```

# Expected behavior:
Exception: out of bounds memory access

wenyongh, 2024-01-02:
Hi, thanks for reporting the issue! Please try #2969.

wenyongh, 2024-01-05:
Close this issue since it was fixed.

closed 2024-01-05 by wenyongh