wasm-micro-runtime/wasm-micro-runtime #2951: Unexpect excrption (integer overflow) on fast JIT mod

erxiaozhou, 2023-12-31:
# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

fast jit mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_FAST_JIT=1 -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1  -Bbuild
```

execute:

fast jit mode
```
iwasm --heap-size=0  --fast-jit -f to_test   <tc_name>
```
# Cases

[iwasm_fast_jit_unexpected_int_overflow.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13801277/iwasm_fast_jit_unexpected_int_overflow.zip)

# Actual behavior:

fast jit mode
```
Exception: integer overflow
```

# Expected behavior:
No exception

wenyongh, 2024-01-02:
Fast JIT works normal, it should be the issue of fast-interp according to the build command and test result.

erxiaozhou, 2024-01-02:
> Fast JIT works normal, it should be the issue of fast-interp according to the build command and test result.

Yes, you are right.

xujuntwt95329, 2024-01-04:
Thanks for reporting this, we've fixed in https://github.com/bytecodealliance/wasm-micro-runtime/pull/2974

wenyongh, 2024-01-05:
Close this issue since it was fixed.

closed 2024-01-05 by wenyongh