wasm-micro-runtime/wasm-micro-runtime #2953: Unexpect excrption (allocate memory failed) on fast interpreter mode

erxiaozhou, 2023-12-31:
# Build commands
I compile the code with commit id 40b430fd240ad45f5c7a7dd0239fa2425e7c294e.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

fast jit mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:

fast jit mode
```
iwasm --heap-size=0  <tc_name>
```
# Cases

[iwasm_fast_interp_unexpected_exception2.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13801304/iwasm_fast_interp_unexpected_exception2.zip)

# Actual behavior:

fast jit mode
```
Exception: allocate memory failed
```

# Expected behavior:
No exception
```
0x16006d52ce3ebd1a:i64
```

xujuntwt95329, 2024-01-04:
Thanks for reporting this, we've fixed in https://github.com/bytecodealliance/wasm-micro-runtime/pull/2974

wenyongh, 2024-01-05:
Close this issue since it was fixed.

closed 2024-01-05 by wenyongh