wasm-micro-runtime/wasm-micro-runtime #2958: Wrong execution behavior of the instruction `i64.gt_s` on fast JIT mode

erxiaozhou, 2023-12-31:
# Build commands
I compile the code with commit id 92bf8547aa64dddad2ecd387a352f89bbc475c88.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_FAST_JIT=1   -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1  -Bbuild
```

execute:

```
iwasm --heap-size=0  --fast-jit -f to_test <tc_name>
```

# Case

[i64.gt_s_42.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13802003/i64.gt_s_42.zip)

# Actual output:

```
iwasm --heap-size=0  --fast-jit -f to_test i64.gt_s_42.wasm
0x0:i32
```

# Expected output:

```
iwasm --heap-size=0 --llvm-jit -f to_test i64.gt_s_42.wasm
0x1:i32
```

wenyongh, 2024-01-02:
Hi, thanks for reporting the issue! Please try https://github.com/bytecodealliance/wasm-micro-runtime/pull/2969.

wenyongh, 2024-01-05:
Close this issue since it was fixed.

closed 2024-01-05 by wenyongh