wasm-micro-runtime/wasm-micro-runtime #2967: build wamr-compiler error

beautiful-boyyy, 2024-01-01:
cd wamr-compiler
./build_llvm.sh
```
[notice] A new release of pip is available: 23.3.1 -> 23.3.2
[notice] To update, run: python3.11 -m pip install --upgrade pip
To use ssh for git clone, run: export USE_GIT_SSH=true
Use the gcc toolchain
CMake Deprecation Warning at CMakeLists.txt:8 (cmake_policy):
  The OLD behavior for policy CMP0114 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```
sys: macos 14.2 arm64

lum1n0us, 2024-01-01:
> CMake was unable to find a build program corresponding to "Ninja".

We use "ninja" to speed up LLVM libraries compilation. And "ninja" is missing on your machine.

beautiful-boyyy, 2024-01-02:
> > CMake was unable to find a build program corresponding to "Ninja".
> 
> We use "ninja" to speed up LLVM libraries compilation. And "ninja" is missing on your machine.

Thanks successfuly build wamr-compiler, but when I want to generate aot files, another problem occurs.
```
Run binarytrees_class benchmark by WAMR aot:
LLVM ERROR: Only small, tiny and large code models are allowed on AArch64
./run.sh: line 46: 49552 Abort trap: 6           $wamrc --enable-gc -o $prefix.aot $prefix.wasm > tmp.txt
Read file to buffer failed: open file binarytrees_class.aot failed.
```

lum1n0us, 2024-01-02:
May I ask the command involve `wamrc` ? especially which command options are using.

beautiful-boyyy, 2024-01-02:
> May I ask the command involve `wamrc` ? especially which command options are using.

#Yes. It's in this file [url](https://github.com/web-devkits/Wasmnizer-ts/blob/main/tests/benchmark/run_benchmark.js)

closed 2024-01-02 by beautiful-boyyy