wasm-micro-runtime/wasm-micro-runtime #2989: Build failure with esp-idf (libc_errno.h missing)

h1romas4, 2024-01-09:
Hello.

I have built WARM (`1.3.0` `78be221`) on esp-idf (`v4.4.6`) and have found that it fails to compile with the following error.

```
/home/hiromasa/devel/esp32/m5stamp-c3wamr/components/wasm/
wasm-micro-runtime/core/shared/platform/esp-idf/espidf_socket.c:8:10: 
fatal error: libc_errno.h: No such file or directory
 #include "libc_errno.h"
          ^~~~~~~~~~~~~~
compilation terminated.
```

`CMakeLists.txt` configuration:

```
# wasm-micro-runtime settings
list(APPEND EXTRA_COMPONENT_DIRS "components/wasm/wasm-micro-runtime/build-scripts/esp-idf")
set(WAMR_BUILD_INTERP 1)
set(WAMR_BUILD_FAST_INTERP 1)
set(WAMR_BUILD_AOT 0)
set(WAMR_BUILD_LIBC_BUILTIN 1)
set(WAMR_BUILD_APP_FRAMEWORK 0)
```

This event does not occur in WARM (`1.2.3` `fa2f29f`), which compiles correctly.
I found this when trying to upgrade the WAMR version from 1.2.3 to 1.3.0.

Best regard.

wenyongh, 2024-01-09:
Hi, could you try the below patch?
```diff
diff --git a/core/shared/platform/esp-idf/shared_platform.cmake b/core/shared/platform/esp-idf/shared_platform.cmake
index 8d7fe5d8..6cd4cc42 100644
--- a/core/shared/platform/esp-idf/shared_platform.cmake
+++ b/core/shared/platform/esp-idf/shared_platform.cmake
@@ -10,6 +10,9 @@ include_directories(${PLATFORM_SHARED_DIR}/../include)

 file (GLOB_RECURSE source_all ${PLATFORM_SHARED_DIR}/*.c)

+include (${CMAKE_CURRENT_LIST_DIR}/../common/libc-util/platform_common_libc_util.cmake)
+set(source_all ${source_all} ${PLATFORM_COMMON_LIBC_UTIL_SOURCE})
+
 set (PLATFORM_SHARED_SOURCE ${source_all} ${PLATFORM_COMMON_MATH_SOURCE})
```

h1romas4, 2024-01-09:
Thank you for your response.

After applying the patch, it now compiles correctly!

```diff
diff --git a/core/shared/platform/esp-idf/shared_platform.cmake b/core/shared/platform/esp-idf/shared_platform.cmake
index 8d7fe5d8..e31c8653 100644
--- a/core/shared/platform/esp-idf/shared_platform.cmake
+++ b/core/shared/platform/esp-idf/shared_platform.cmake
@@ -8,6 +8,9 @@ add_definitions(-DBH_PLATFORM_ESP_IDF)
 include_directories(${PLATFORM_SHARED_DIR})
 include_directories(${PLATFORM_SHARED_DIR}/../include)
 
+include (${CMAKE_CURRENT_LIST_DIR}/../common/libc-util/platform_common_libc_util.cmake)
+set(source_all ${source_all} ${PLATFORM_COMMON_LIBC_UTIL_SOURCE})
+
 file (GLOB_RECURSE source_all ${PLATFORM_SHARED_DIR}/*.c)
 
 set (PLATFORM_SHARED_SOURCE ${source_all} ${PLATFORM_COMMON_MATH_SOURCE})
```

```
esptool.py v3.3.4-dev
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.
Generated /home/hiromasa/devel/esp32/m5stamp-c3wamr/build/m5stamp-c3wamr.bin
```

h1romas4, 2024-01-09:
I appreciate your help!

wenyongh, 2024-01-11:
Welcome. Close the issue as it was fixed.

closed 2024-01-11 by wenyongh