wasm-micro-runtime/wasm-micro-runtime #3009: Unable to run iwasm on MacOS after #2992

eloparco, 2024-01-12:
After #2992, I'm not able to run `iwasm` on MacOS.

When I build, I get
```bash
ld: warning: non-standard -pagezero_size is deprecated when targeting macOS 13.0 or later
```

And if I run `iwasm` I get:
```bash
./iwasm
[1]    2986 killed     ./iwasm
```

If I try running with a debugger, I get
```bash
lldb -- ./iwasm                               
(lldb) target create "./iwasm"
Current executable set to '/Users/eloparco/dev/forks/wasm-micro-runtime/samples/wasi-threads/build/iwasm' (arm64).
(lldb) run
error: Malformed Mach-o file
```

If I comment this line out
https://github.com/bytecodealliance/wasm-micro-runtime/blob/08442458b12244f2cacb2f49573bb84e9433071f/build-scripts/config_common.cmake#L469
everything is fine.

wenyongh, 2024-01-15:
@eloparco, @yamt can we add some checks for this flag? For example, can we only enable it when the macOS version is smaller than 13.0?

yamt, 2024-01-15:
i tested https://github.com/bytecodealliance/wasm-micro-runtime/pull/2992 with macOS/x86-64.
```
ProductName:    macOS
ProductVersion: 12.7.1
BuildVersion:   21G920
```
i don't have other versions of macOS handy.

eloparco, 2024-01-15:
I'm using:
```bash
$ sw_vers
ProductName:            macOS
ProductVersion:         13.3.1
BuildVersion:           22E261
```

yamt, 2024-01-15:
> I'm using:
> 
> ```shell
> $ sw_vers
> ProductName:            macOS
> ProductVersion:         13.3.1
> BuildVersion:           22E261
> ```

does `man ld` on your system have some info about the deprecation?
mine has:
```
     -pagezero_size size
             By default the linker creates an unreadable segment starting at
             address zero named __PAGEZERO.  Its existence will cause a bus
             error if a NULL pointer is dereferenced.  The argument size is a
             hexadecimal number with an optional leading 0x.  If size is zero,
             the linker will not generate a page zero segment.  By default on
             32-bit architectures the page zero size is 4KB.  On 64-bit
             architectures, the default size is 4GB.
```

eloparco, 2024-01-15:
>does man ld on your system have some info about the deprecation?

No, it's the same as yours

yamt, 2024-01-16:
> And if I run `iwasm` I get:
> 
> ```shell
> ./iwasm
> [1]    2986 killed     ./iwasm
> ```
> 
> If I try running with a debugger, I get
> 
> ```shell
> lldb -- ./iwasm                               
> (lldb) target create "./iwasm"
> Current executable set to '/Users/eloparco/dev/forks/wasm-micro-runtime/samples/wasi-threads/build/iwasm' (arm64).
> (lldb) run
> error: Malformed Mach-o file
> ```

does it mean, a binary built for 12 doesn't work on 13?
("cmake -DCMAKE_OSX_DEPLOYMENT_TARGET=12.0 ..")

wenyongh, 2024-01-16:
It is strange that the issue was closed automatically after #3025 was merged. I re-open it again.

yamt, 2024-01-16:
cf. https://github.com/golang/go/issues/54482
they just removed the option as it was not actually useful for them. (unlike us)

eloparco, 2024-01-16:
>does it mean, a binary built for 12 doesn't work on 13?
("cmake -DCMAKE_OSX_DEPLOYMENT_TARGET=12.0 ..")

Yes, so we should probably do this https://github.com/bytecodealliance/wasm-micro-runtime/pull/3025#discussion_r1453121897 and in the release CI we build with `-DCMAKE_OSX_DEPLOYMENT_TARGET=13.0`

yamt, 2024-01-16:
i'm inclined to revert https://github.com/bytecodealliance/wasm-micro-runtime/pull/2992 as it's cumbersome to deal with incompatibilities among osx versions.

PR #3025 (2024-01-16): Do not use pagezero size option if osx version >= 13
Fixes #3009
closed 2024-02-14 by eloparco