wasm-micro-runtime/wasm-micro-runtime #3013: the `memory_instantiate` method on iOS fails when executing `os_mmap`.

panicfrog, 2024-01-15:
Beacause `map_size` is 8G. I modified it to 2GB and it works fine on my iPhone 13 mini. I guess it has something to do with the iPhone's RAM. iPhone x only has 3G RAM. only iPhone 15 and iPhone 15 pro have 8G RAM. Most devices only have 4G or 6G RAM.
[https://9to5mac.com/2023/11/18/iphone-ram-list/](iphone ram list)

wenyongh, 2024-01-15:
Hi, if os_mmap 8G is not supported, maybe you can try disabling the hardware boundary check feature when building iwasm:
`cmake -DWAMR_BUILD_HW_BOUND_CHECK=0`
And use `wamrc --bounds-checks=1` to generate AOT file with software boundary check enabled instead. Refer to:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#disable-boundary-check-with-hardware-trap

panicfrog, 2024-01-16:
> Hi, if os_mmap 8G is not supported, maybe you can try disabling the hardware boundary check feature when building iwasm: `cmake -DWAMR_BUILD_HW_BOUND_CHECK=0` And use `wamrc --bounds-checks=1` to [...]

The `pthread_jit_write_protect_np` method is not allowed to be used on iOS, because the use of jit is prohibited on iOS. So there should be no way to use aot and jit modes on iOS.

wenyongh, 2024-01-16:
If AOT/JIT isn't allowed in iOS, please build iwasm with `cmake -DWAMR_DISABLE_HW_BOUND_CHECK=1`, it will disable hardware boundary check (which os_mmap 8G memory) for interpreter, and enable the software boundary check for interpreter automatically.

panicfrog, 2024-01-18:
> If AOT/JIT isn't allowed in iOS, please build iwasm with `cmake -DWAMR_DISABLE_HW_BOUND_CHECK=1`, it will disable hardware boundary check (which os_mmap 8G memory) for interpreter, and enable the software boundary check for interpreter automatically.

Thank you very much for your reply, I followed your reply and it works fine. I still have a question: What is the difference between `hardware boundary check` and `software boundary check`? What will be the impact?

wenyongh, 2024-01-18:
Welcome, hardware boundary check means using the hardware trap mechanism to do the wasm memory access boundary check, some guard pages are pre-allocated and will be accessed when the load/store address of wasm memory operation is out of range, which raises trap and triggers signal handler to throw the exception. So in the interpreter or aot/jit code, the checking instructions can be removed. And for the software boundary check, they are added. And you can see it impacts the performance, the former is better as some check instructions are saved.

closed 2024-01-18 by panicfrog