wasm-micro-runtime/wasm-micro-runtime #3023: Unexpected behavior of the instruction `table.init`

erxiaozhou, 2024-01-15:
# Introduction
I executed the test case containing a `table.init` instruction and found that warm failed to raise an exception as expected. 

# Build commands
I compile the code with commit id 892a94fd05f2e95fc68535d763bf67ab1b8e6263.
Platform: Ubuntu 20.04
CPU: amd64

Here we take the warm on JIT mode for example. This issue also exists in warm on other modes (classic interpreter mode, fast interpreter mode, fast JIT mode, multi JIT mode).
 
compile:
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_FAST_JIT=1   -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1  -Bbuild
```

```
iwasm --heap-size=0  --fast-jit -f to_test <tc_name>
```

# Case

[all_wamr_table.init_no_exception.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13941943/all_wamr_table.init_no_exception.zip)

# Actual output:
No exception

# Expected output:
There is an exception indicating "out of bounds table access".

erxiaozhou, 2024-01-16:
Let me double-check this issue ... Please bear with me for a moment ...

wenyongh, 2024-01-16:
@erxiaozhou don't worry, it's nice to spot this issue. I think it should be similar to issue https://github.com/bytecodealliance/wasm-micro-runtime/issues/3020, the src offset of table segment is 1 (s = 0), the length is 0 (n = 0), and the length of table segment is (len(table.data) = 1), though `s+n` isn't larger than `len(table.data)`, but `s` is out of range. It seems it doesn't violate the spec, but some runtimes add the check.

erxiaozhou, 2024-01-16:
> @erxiaozhou don't worry, it's nice to spot this issue. I think it should be similar to issue #3020, the src offset of table segment is 1 (s = 0), the length is 0 (n = 0), and the length of table [...]

Thank you for your patience and nice reply! This difference is really interesting.

closed 2024-01-17 by erxiaozhou