wasm-micro-runtime/wasm-micro-runtime #3024: Is it ok to call directly into the aot file?

LHolten, 2024-01-15:
I am trying to time exactly how long it takes to execute some piece of wasm on a microcontroller (esp32c3).
The problem is that `wasm_runtime_call_wasm` does a lot of work checking that the arguments are correct etc.
This method is also not in RAM which means that there is caching which makes timing inconsistent.

I have figured out that I can retrieve the function pointer to the actual AOT code in RAM from inside `AOTFunctionInstance`.
Calling this function with the execution environment and arguments seems to work perfectly.
Assuming that I have checked that the function signature is correct etc, is this correct?

wenyongh, 2024-01-16:
Hi, normally it should be good as long as the software boundary check is enabled by wamrc and hw bound check is disabled when compiling iwasm, it seems that they are in the right settings by default since the esp32c3 is 32-bit. And had better check whether exception was thrown with `wasm_runtime_set_exception` after calling the AOT function.

LHolten, 2024-01-17:
Thank you, this is really cool!