wasm-micro-runtime/wasm-micro-runtime #3026: Unexpected exception on fast interpreter mode

erxiaozhou, 2024-01-15:
# Build commands
I compile the code with commit id 892a94fd05f2e95fc68535d763bf67ab1b8e6263.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:

```
iwasm --heap-size=0  <tc_name>
```
# Cases
[wamr_fast_interpreter_memory_oob.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13942052/wamr_fast_interpreter_memory_oob.zip)

# Actual output:

```
Exception: out of bounds memory access
```

# Expected output:

```
0xffffffffffffffff:i64
```

wenyongh, 2024-01-22:
@erxiaozhou thanks for reporting the issue, it was fixed with https://github.com/bytecodealliance/wasm-micro-runtime/pull/3058.

erxiaozhou, 2024-01-24:
Thank you for your reply!

closed 2024-01-24 by erxiaozhou