wasm-micro-runtime/wasm-micro-runtime #3027: Unexpect exception (memory OOB) on JIT mode

erxiaozhou, 2024-01-15:
# Build commands
I compile the code with commit id 892a94fd05f2e95fc68535d763bf67ab1b8e6263.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

jit mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=0 -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1 -Bbuild
```

execute:

```
iwasm --heap-size=0 --llvm-jit -f to_test  <tc_name>
```
# Cases
[wamr_jit_memory_oob2.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13942116/wamr_jit_memory_oob2.zip)

# Actual behavior:
```
Exception: out of bounds memory access
```

# Expected behavior:
No exception

lum1n0us, 2024-01-17:
Thanks for catching it. 

It is a known limitation about the ***variant representation of NaN***. We tend to keep it.

As you know, an IEEE754 single precision NaN would be encoded as `s111 1111 1xxx xxxx xxxx xxxx xxxx xxxx` where `s` is the sign and `x` represents a non-zero number. In practice, the `x` sequence, called *payload*, is most often ignored in applications. As a result, `0xffffffff` and `0x7fc00000` both are `NaN`.

In this case, there is an op `f64x2.max` with two v128 operands *0xd46d8373 0x33b70bcd 0xca3508a1 0x38fa9332* and *0xffffffff 0xffffffff 0xffffffff 0xffffffff*. The second operand is exactly *NaN NaN*. The result will be *NaN NaN* according to the Wasm spec. Variant runtimes will encode the result in different ways. Some runtime will use qNaN, which means the payload is all zeros. Others, including WAMR, will use the operand itself(0xffffffff 0xffffffff).  Both are technically right. 

Then the result, *NaN NaN* will be used as an operand of the following op `i64.load16_s` to represent the memory address. Eventually, it leads to an OOB exception.

erxiaozhou, 2024-01-17:
Get it! Thank you for you reply!

closed 2024-01-17 by erxiaozhou