wasm-micro-runtime/wasm-micro-runtime #3038: [homebrew] wasm-micro-runtime 1.3.1 regression test failed with arm builds

chenrui333, 2024-01-16:
👋 While upgrading the formula to the latest release, 1.3.1, we only saw the regression test failures with the arm builds, need some help figuring out why. Thanks! 🙏 

```
  ==> /opt/homebrew/Cellar/wasm-micro-runtime/1.3.1/bin/iwasm -f fib /private/tmp/wasm-micro-runtime-test-20240112-6128-q7wa56/fib.c.wasm 2>&1
  sh: line 1:  6139 Killed: 9               /opt/homebrew/Cellar/wasm-micro-runtime/1.3.1/bin/iwasm -f fib /private/tmp/wasm-micro-runtime-test-20240112-6128-q7wa56/fib.c.wasm 2>&1
  
  Error: wasm-micro-runtime: failed
  An exception occurred within a child process:
    Minitest::Assertion: Expected: 1
    Actual: 137
```

relates to https://github.com/Homebrew/homebrew-core/pull/159778

wenyongh, 2024-01-17:
Hi, thanks for reporting the issue, would you mind telling us how the iwasm binary is built? Could you upload some logs of build? And it is a little strange that you run the fib case without argument, I remember normally the argument is added, like `iwasm -f fib fib.c.wasm 10`.

tonibofarull, 2024-01-17:
I tested building in linux with `docker run --platform linux/arm64 ubuntu:22.04` and I got no error with `iwasm -f fib fib.c.wasm 1`.

yamt, 2024-01-19:
> 👋 While upgrading the formula to the latest release, 1.3.1, we only saw the regression test failures with the arm builds, need some help figuring out why. Thanks! 🙏
> 
> ```
>   ==> /opt/homebrew/Cellar/wasm-micro-runtime/1.3.1/bin/iwasm -f fib /private/tmp/wasm-micro-runtime-test-20240112-6128-q7wa56/fib.c.wasm 2>&1
>   sh: line 1:  6139 Killed: 9               /opt/homebrew/Cellar/wasm-micro-runtime/1.3.1/bin/iwasm -f fib /private/tmp/wasm-micro-runtime-test-20240112-6128-q7wa56/fib.c.wasm 2>&1
>   
>   Error: wasm-micro-runtime: failed
>   An exception occurred within a child process:
>     Minitest::Assertion: Expected: 1
>     Actual: 137
> ```
> 
> relates to [Homebrew/homebrew-core#159778](https://github.com/Homebrew/homebrew-core/pull/159778)

first, i don't understand the intention of the test.
it seems that shell_output() here asserts the exit code is 1.
why do you expect exit code 1 when running this fib.c.wasm?

second, exit code 137 usually means the process was terminated by SIGKILL.
as far as i know, wamr doesn't use SIGKILL by itself.
i suspect it can be an environment issue.

chenrui333, 2024-01-28:
> first, i don't understand the intention of the test. it seems that shell_output() here asserts the exit code is 1. why do you expect exit code 1 when running this fib.c.wasm?
>
> second, exit code 137 [...]

yeah, expecting error code as 1. 

yeah, I understand the error code meaning, but this is regression test failure (so that is why I dumped the test results in here)

closed 2024-02-02 by chenrui333