wasm-micro-runtime/wasm-micro-runtime #3050: Handling "Exception: wasi proc exit"

nchamzn, 2024-01-18:
Hi, 

After [this change](https://github.com/bytecodealliance/wasm-micro-runtime/pull/2529), is an embedder expected to check the exception matches "Exception: wasi proc exit" as iwasm now does to validate if there was an error? How stable can we expect that string to be?

Thanks.

wenyongh, 2024-01-19:
Hi, "wasi proc exit" is a normal behavior and the exception will be cleared by runtime before wasi app main function returns.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/8b37048823677d9229b7458b064a515b19744c9c/core/iwasm/common/wasm_application.c#L132-L137
Please use `wasm_runtime_get_exception` to check whether there was exception thrown.

nchamzn, 2024-01-19:
Hi, sorry - I meant as an embedder so not using iwasm. I think at the moment if we call a wasm function then we are required to check for an exception and then ignore it if it matches "Exception: wasi proc exit"?

wenyongh, 2024-01-19:
Hi, yes, the "wasi proc exit" is only thrown when the wasm app is a wasi module, and it is cleared by runtime if the function to call is "_start" function and developer uses wasm_application_execute_main to call it. If developer calls other functions or calls "_start" function without using wasm_application_execute_main, normally he should ignore the "wasi proc exit" by himself.