wasm-micro-runtime/wasm-micro-runtime #3061: Crash in fast interpreter

erxiaozhou, 2024-01-20:
# Build commands
I compile the code with commit id 1977ad23efb3b3f2d4ae0ddf49096a05464f488f.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:

```
iwasm --heap-size=0  <tc_name>
```
# Cases
[wamr_fast_interpreter_crash.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13998829/wamr_fast_interpreter_crash.zip)

(I have shrunken the case, but it is still a little long ...)

# Actual output:

```
double free or corruption (out)
Aborted
```

# Expected output:

```
1
```

wenyongh, 2024-02-02:
@erxiaozhou thanks for spotting the issue, it was fixed with #3112, please pull the latest code to test again.

wenyongh, 2024-03-15:
Close this issue since it had been resolved.

closed 2024-03-15 by wenyongh