wasm-micro-runtime/wasm-micro-runtime #3062: Wrong execution result in fast interpreter

erxiaozhou, 2024-01-21:
# Build commands
I compile the code with commit id 1977ad23efb3b3f2d4ae0ddf49096a05464f488f.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:

```
iwasm --heap-size=0  <tc_name>
```
# Case
[wamr_fast_interpreter_wrong_float.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/14001616/wamr_fast_interpreter_wrong_float.zip)

(I have shrunken the case, but it is still a little long ...)

## Code in WAT
```
(module
  (type (;0;) (func (result f32)))
  (type (;1;) (func (param f32) (result f32)))
  (func (;0;) (type 0) (result f32)
    f32.const 0x1.cd9d78p-90 (;=1.4566e-27;)
    call 1
    i32.const 1
    if  ;; label = @1
      nop
    else
      f32.const -0x1.756784p+89 (;=-9.02836e+26;)
      return
      i32.const 1
      i32.le_s
      f32.convert_i32_u
      call 1
      global.set 0
    end)
  (func (;1;) (type 1) (param f32) (result f32)
    i32.const 0
    if (result f32)  ;; label = @1
      f32.const 0x0p+0 (;=0;)
    else
      f32.const nan (;=nan;)
    end)
  (global (;0;) (mut f32) (f32.const -0x1.c9cf4ep+103 (;=-1.81357e+31;)))
  (export "to_test" (func 0)))
```

# Actual output:

```
0
```

# Expected output:

```
nan
```

erxiaozhou, 2024-01-24:
It seems that this issue and https://github.com/bytecodealliance/wasm-micro-runtime/issues/3061 suffer from the same root cause.

wenyongh, 2024-02-02:
@erxiaozhou thanks for spotting the issue, it was fixed with https://github.com/bytecodealliance/wasm-micro-runtime/pull/3101, please pull the latest code to test again.

wenyongh, 2024-03-15:
Close this issue since it had been resolved.

closed 2024-03-15 by wenyongh